moonglow/FlashForge_Marlin #3: Dreamer Nx issues/bugs/concerns

tckb, 2021-04-03:
Thanks for the instructions! the installation was flawless and went smooth. Here's the pic, 

[image]

I have noticed some quirks, and possibly bugs. I will use this issue for documenting it.  I will keep it open until all the issues are fixed or will be fixed so that others can keep track of it. 

Open issues:
- [ ]

moonglow, 2021-04-03:
Can you provide me more information about homing issue ?

tckb, 2021-04-03:
3# oscillating positions on startup.

https://user-images.githubusercontent.com/939542/113492974-d2c7f300-94db-11eb-9b48-724e93fd45dd.mov

tckb, 2021-04-03:
2. Homing issue

https://user-images.githubusercontent.com/939542/113493395-3a337200-94df-11eb-95e5-3803928faae4.mp4

moonglow, 2021-04-04:
1. Marlin does not have support for PCA9632 PWM controller as caselight ( very strange because LED class has it and Caselight logic use LED class to operate with it ) , but i think i can fix it
2. Where is no issue with it, it is just controlled back-off after homing, after homing switches was reached and retested with less speed, i set `#define Z_AFTER_HOMING  20` and `#define HOMING_BACKOFF_POST_MM { 10, 10, 0 }` for XY
3. It is standard Marlin behavior it says to user what "printer does not know real position ( no homing procedure was ended ) be careful" after homing procedure was ended, position will stop blinks too

tckb, 2021-04-04:
>     1. Marlin does not have support for PCA9632 PWM controller as caselight ( very strange because LED class has it and Caselight logic use LED class to operate with it ) , but i think i can fix it

That would be awesome! having the light turned on would really be cool. The light switch control is pretty deeply nested inside to switch on.

>     2. Where is no issue with it, it is just controlled back-off after homing, after homing switches was reached and retested with less speed, i set `#define Z_AFTER_HOMING  20` and `#define HOMING_BACKOFF_POST_MM { 10, 10, 0 }` for XY

Ah! I get it now.  This was different than the stock firmware where there was no backoff, is the backoff neccessary? at least due the xy backoff, the hotend seemed to be _jerking_ after homing. 

>     3. It is standard Marlin behavior it says to user what "printer does not know real position ( no homing procedure was ended ) be careful" after homing procedure was ended, position will stop blinks too

Yes, you are right. The blinking stops as soon as there is some moment.

moonglow, 2021-04-04:
>backoff neccessary

i remove it from XY in v0.3, but Z back-off is useful for me to observe and clean surface while print was started.

>Can not load or unload filament, the related gcode doesn't work

Because it is part of advanced pause feature, now it supported in v0.3

>How to access the internal memory?

Internal SD card **will not be accessible** and we have two reason for that:
1. Marlin support only one SD card ( external is better for this )
2. I do not want to allow modify anything what will prevent user to revert firmware to original one whats why i also do not use onboard EEPROM memory.
>WiFi, not sure if this is supported in Marlin,. there's no option of Wifi

Marlin supports WiFi but not support WiFi module from FlashForge :) i think much easy way to use some cheap WiFi module esp8266 based and use standard marlin WiFi driver for what. It is now in my very far TODO list.

Now chamber light will ON by default in version 0.3

tckb, 2021-04-04:
@moonglow with v0.3 filament loading and unloading is possible but this is still a bit tricky. This could be a Marlin specific stuff.  Also the "z" only backoff works perfect! thanks. 

>I do not want to allow modify anything what will prevent user to revert firmware to original one whats why i also do not use onboard EEPROM memory.

I think for beta version, this is a good decision! 

>Marlin support only one SD card ( external is better for this )

 I see, it would be awesome if we could leverage the disk. 
 
 >Marlin supports WiFi but not support WiFi module from FlashForge :) 
 
 Thats a bummer!  I am not in a strict use for Wifi as I already have an octopi attached.  I think it would be good to add a note for this,.