moonglow/FlashForge_Marlin #5: Enable Power-loss Recovery

tckb, 2021-04-04:
Now that we are out Flashforge, we can enable support for Power-loss recovery.  Although, there might not be a full fledged support for power loss pin, marlin still supports a  sd card based recovery albeit discouraged (in the docs) I think it should be beneficial. 

config: `POWER_LOSS_RECOVERY`

ref: https://marlinfw.org/docs/gcode/M413.html

personally, I would like it as they can disabled and enabled again when needed.

moonglow, 2021-04-04:
It enabled in Marlin_Inventor firmware because they use Z home at bottom ( Z-max endstop ) and after powerloss you can home again and restart printing, in our printers we can't goes to HOME position because we smash our model ( but we can disable "goes to home" on power-loss recovery ) and all looks ok ? but our table without power can goes lower... and in most cases your print be ruined on power-loss resume. But i can enabled it too

tckb, 2021-04-04:
@moonglow  You are absolutely right. I never thought about it. But Yes, if the "goes home" can be disabled that'd be ideal. 
In an ideal world,  I would imagine  this scenario:
```
at some "z" height, there's power-loss, the build plate slips down (i.e goes to max z). after the power on, detected print failure, shows a failure message on LCD and asks to retry, goes back to last known position of "z" failure and bam print continues.
```
can you imagine ☝️  scenario?

moonglow, 2021-04-04:
You your build-plate goes down you can't goes to last know position because you can't know the current position without homing :( IMHO Better to change placement of Z end-stop ( also connect it to Z-max socket ) as it was in Inventor and use marlin_Inventor firmware, you also will get more height for building and better power-loss recovery

tckb, 2021-04-05:
>you can't know the current position without homing

right, I realized it, a negative homing works :P 

> IMHO Better to change placement of Z end-stop

that'd be tinkering with the hardware. this is not a novice job. I can do it but, do not want to tinker for the software to work.

closed 2021-04-09 by moonglow