moonglow/FlashForge_Marlin #6: Motor current control

moonglow, 2021-04-04:
Marlin current setup via M907 use real current values, as example default values are:

`M907 X0.8 Y0.8 Z0.5 E0.6 B0.6`

FlashForge use raw value ( from 0 to 127 ), which is better ?  maybe change it to FlashForge behavior ?

tckb, 2021-04-04:
IMO, the raw real values has a  better resolution than  using 7 bits.

moonglow, 2021-04-04:
MCP4018 digital potentiometer has 7 bit resolution ( 0 - 127 ) but you need to calculating yourself needed current ,now i calculate it in firmware from Amperes to 0-127 in firmware

tckb, 2021-04-04:
I see! but I think to set the expectations (read "comply") from the doku, the real current values should be used, even though the actual values are computed. I would expect to see a note.

closed 2021-04-09 by moonglow