moonglow/FlashForge_Marlin #7: Enable Bed leveling?

tckb, 2021-04-04:
I am not sure if bed leveling is enabled, G29 just says 

```
Send: G29
Recv: Mesh Bed Leveling has no data.
Recv: X:270.00 Y:150.00 Z:20.00 E:0.00 Count X:24003 Y:13335 Z:8000
Recv: ok

```

Bed leveling in the menu just "homes" and doesn't start the leveling.  I would like to test this "Mesh" leveling feature in Marlin2.x 

as per: 
https://marlinfw.org/docs/gcode/G029-mbl.html

requires the following to the enabled 

```
MESH_BED_LEVELING
LCD_BED_LEVELING
```
NB: 
"Level Corner" triggers leveling the corners by selecting multiple corner points..

moonglow, 2021-04-04:
Original firmware use Level Corner, you want to use bed leveling without sensor and manually goes to all this dots ? But i will enable it, no problem

tckb, 2021-04-04:
Yes, you are correct. But, I would like to use the gcode for leveling and as with others its deeply nested inside the  menu for this Level corner. Is there a gcode for this `level corner` ?

I'd still want to use the mesh-leveling, just for the seeing if it has any improvements.

moonglow, 2021-04-04:
> Is there a gcode for this level corner ?
i think where is no G-Code for such things,but you can write it manually

moonglow, 2021-04-05:
enabled in v0.4

tckb, 2021-04-05:
Yup, it works in v0.4. Although for nx, three points where in "air" the rest of the leveling works. I will do a test print to see if it works fine.

closed 2021-04-05 by tckb