moonglow/FlashForge_Marlin #9: Printing issues

tckb, 2021-04-05:
I am not able print anything. After issues "Upload & Print" via Octoprint, the homing works but, the printhead makes a terribly slow movement to the build plate and starts printing _right away_ without waiting for the correct temp to reach. Thankfully, octoprint blocks the extrusion 

```
Send: G1 X13.569 Y15.167 E6.42610
Recv: echo: cold extrusion prevented
Recv: ok
```
Attaching the test file - 
[sunglass_holderv3_PLA_30m.gcode.txt](https://github.com/moonglow/FlashForge_Marlin/files/6258475/sunglass_holderv3_PLA_30m.gcode.txt)

@moonglow  can you test it ?

moonglow, 2021-04-05:
I think error is very literally wants to say you, what you want to extrude plastic on cold extruder. `M140` and ` M104` will not wait for temperature you need you some generic G-codes startup script. I also notice you want to use some FlashForge specific G-Codes as example `M132` `G161`  `M6`  and invalid motor current settings too `M907 X100 Y100 Z40 A80 B20` - it will set maximum possible current to drivers o_O If you prefer Prusa slicer you can use any generic printers profile as start point as example Ender 3 profile or something like that. FlashForge guys use the own and some times very specific logic what unsupported by mostly all other slicers

moonglow, 2021-04-05:
As example i made this test cube in few minutes from scratch with PrusaSlicer, you can check with it.

[xyzCalibration_cube.gcode.txt](https://github.com/moonglow/FlashForge_Marlin/files/6258652/xyzCalibration_cube.gcode.txt)

tckb, 2021-04-05:
@moonglow  Yes I thought it might be because of that. I use the following gcode for start and end -- this was compiled by someone else. I am not very aware of this. May be you can fix this?

start g-code 
```
;start G code
;Machine set up

M140 S[first_layer_bed_temperature] T0;   Heat bed up to first layer temperature
M104 S[first_layer_temperature_0] T0;   Set nozzle temperature to first layer temperature
M107 ; fan off
G90 ; absolute positioning
G28 ; home
M132 X Y Z A B; load axis offsets from eeprom
G1 Z50.00 F400 ; Lower bed
G161 X Y F3300 ; home x and y to min
M6 T0 ; select tool 0
M907 X100 Y100 Z40 A80 B20 ; Set motor voltages
M108 T0 ; Tool 0

;Nozzle purge move

G1 X-70 Y-70 Z0.3 ;Move to bottom left
G1 X70 E20 F1500 ; Extrude across bed
G1 Y-69.6 ; Move up 0.4
G1 X-40 E20 F1500 ; part way back extruding
G1 X-70 F2000 ; Drain nozzle
G1 Z5 ; lift head
```

end g-code 

```
;End G Code
M104 S0 T0 ; heaters off
G28 X Y ; Home x and y
M18 ; disable stepper motors
```

tckb, 2021-04-05:
These works perfectly for flashforge firmware.

moonglow, 2021-04-05:
>These works perfectly for flashforge firmware.

Main problem what it really works _ONLY_ on flashforge firmware and nothing else and Marlin is absolutely different firmware.

Try to configure PrusaSlicer and setup it to works with Marlin based firmware as base config you can use Ender-3 or something similar. I do not test such G-codes but i think it will works.

Start:
```
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
M107
G90 ; use absolute coordinates
M83 ; extruder relative mode
G28 ; home all
```

End:

```
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F600{endif} ; Move print bed down
G1 X50 Y50 F{travel_speed*60} ; present print
{if max_layer_z < max_print_height-10}G1 Z{z_offset+max_print_height-10} F600{endif} ; Move print bed down further down
M140 S0 ; turn off heatbed
M104 S0 ; turn off temperature
M107 ; turn off fan
M84 X Y E ; disable motors
```

tckb, 2021-04-05:
@moonglow  there certainly seems to be a something off with the configuration.  with the above gcode, my print starts setting proper temp etc but, the print is always on the front left corner. I have no idea why, im still investigating this. 

while your gcode does print correctly at the center, the octoprint however, places the print at the corner -- 

[image]

moonglow, 2021-04-05:
i also think what something wrong with configuration, i already print with this firmware on my NX without problem but i prefer print from SD card ( i also did a print via USB from Repetier-Host )

tckb, 2021-04-05:
@moonglow  may be this is the issue

I see info on homing. I did not notice this is before :P 

For dreamer nx - the build volume is  230x150x140 (x,y,z)

[image]

moonglow, 2021-04-05:
>may be this is the issue

i do not think so...

>z is always backoff by 20

yep, because i set back off by 20 by default, we already talk about it before and remove it from XY.

>I see info on homing. I did not notice this is before :P

i think your G-Code use `M117` to write message to LCD, please try to use something predefined like Ender  and so on... do not use configs from FlashForge printer because it absolutely useless with Marlin.

tckb, 2021-04-05:
>i think your G-Code use M117 to write message to LCD, please try to use something predefined like Ender and so on... do not use configs from FlashForge printer because it absolutely useless with Marlin.

Not using it anymore, I am building a new one but, the home max offsets are incorrect.  I started home from menu, without any gcode.

moonglow, 2021-04-05:
I'm sorry but i can't understand what's a problem, i'm not to familiar with modern 3d printing software and use a classic stl + slicer way to make a print. All things like homing, printing, heating is works fine for me, of course  in my level of understanding. Many 3d printers use Marlin as main firmware and if all what you want works with them it must works here too.

tckb, 2021-04-05:
@moonglow  apologies for the confusion, I naively assumed   that "homing"  should indicate the max build volume i.,e x should 230 and y should 150 etc. But this is not the case. I will do some more tests

moonglow, 2021-04-05:
ah... nope, XYZ is always current position and build volume is smaller then actual extruder position in most cases. As example:
You buildplate located from 0 to 230mm, but your extruder can goes right to 270mm to his MAX X limit. But your slicer limits must set to 230

tckb, 2021-04-05:
@moonglow  I figured the issue was with the origin. With Marlin firmware, the origin needs to be at 0,0. I'm using center of the bed as the origin. This somehow is a bit strange but nothing to do with the firmware. I have adjusted the gcode accordingly. closing this issue as it is not a firmware fix.

closed 2021-04-05 by tckb