moonglow/FlashForge_Marlin #12: Enable Linear advance

tckb, 2021-04-06:
It looks like linear advance is disabled. This can be enabled by using 

`LIN_ADVANCE` in  `configuration_adv.h`

You can set the default `k` value to 0

More info here:   https://marlinfw.org/docs/features/lin_advance.html#saving-the-k-factor-in-the-firmware
ref: https://marlinfw.org/docs/gcode/M900.html

moonglow, 2021-04-07:
Yep it is nice feature for advanced users, but `LIN_ADVANCE` and `S_CURVE_ACCELERATION` can't be used together ( even if you set `K` to 0 ) ... if we need to use it together i need to enable also `EXPERIMENTAL_SCURVE` or disable `S_CURVE_ACCELERATION` option :new_moon_with_face:  i can also add _LA postfix fos such firmwares

tckb, 2021-04-07:
I am not too much aware of `EXPERIMENTAL_SCURVE`  would that cause any issues if enabled? also no idea about `S_CURVE_ACCELERATION`  either.  I've seen LIN_ADVANCE has many advantages and wanted to tested it out, esp one of the "key" feature of Marlin 2.0

moonglow, 2021-04-07:
>would that cause any issues if enabled?

it marks as experimental, so i do not know :)

>also no idea about S_CURVE_ACCELERATION

This video shows difference between standard ( trapezoid ) and s-curve acceleration: [LINK](https://www.youtube.com/watch?v=C0XjXqO6Ji8)

This version with LA without S CURVE acceleration for you:
[marlin_nx_la.zip](https://github.com/moonglow/FlashForge_Marlin/files/6270400/marlin_nx_la.zip)

tckb, 2021-04-07:
That is definitely interesting..  since  is this enabled already in v0.4? how do you check or use this? 

there are some clues here
https://github.com/MarlinFirmware/Marlin/issues/14728#issuecomment-616820556

some say that lin_advance with classic acceleration gives better results for  direct extrusion? I am not sure.

moonglow, 2021-04-07:
> since is this enabled already in v0.4? how do you check or use this?

Yep it was enabled already and i think only your feelings can helps, if you like how it moves what mean it works well for you :)

>some say that lin_advance with classic acceleration gives better results for direct extrusion? I am not sure.

Marlin will not allows you to build firmware with S CURVE and LA enabled, whats why they add EXPERIMENTAL CURVE to make possible build it together. marlin_nx_la.zip - uses classic acceleration and LA

tckb, 2021-04-07:
@moonglow 

>i think only your feelings can helps, if you like how it moves what mean it works well for you :)

not sure if I noticed any difference big difference, since I don't print with high speeds.  But, I have some time noticed the hotend moving slowly and making weird sound. :P

>marlin_nx_la.zip - uses classic acceleration and LA

 I will test "marlin_nx_la" in the mean time, but it be great benefit if this is included in the firmware as it vastly increases the print quality at the edges.

moonglow, 2021-04-07:
>it vastly increases the print quality at the edges.

only if someone will calibrate it =) BTW try it without calibration to make sure you like how printers works without S CURVE acceleration, but i think it will be added

tckb, 2021-04-08:
oh hey @moonglow  I asked folks at Marlin, which one they'd choose b/w LA and S-Curve, here's what they said:

>You can enable EXPERIMENTAL_SCURVE to combine them, and maybe it'll work. It comes down to the results you get from experimentation and tuning. You may find the ringing reduction from S-curve is not significant enough to justify the extra computation.

https://twitter.com/MarlinFirmware/status/1379900285930340360

moonglow, 2021-04-08:
> and maybe it'll work

"maybe" hehehe. No problem i will enable SA for experimental build ( also include new parking position and parking on pause )
[marlin_dreamer_nx_la_sa.zip](https://github.com/moonglow/FlashForge_Marlin/files/6276966/marlin_dreamer_nx_la_sa.zip)

tckb, 2021-04-19:
This is already enabled in the 0.7

closed 2021-04-19 by tckb