moonglow/FlashForge_Marlin #13: Bed origin position

moonglow, 2021-04-07:
FlashForge use center of bed as X:0, Y:0 position, i think we need do same with marlin firmware

moonglow, 2021-04-07:
home position now equal to original FF home position and bed center at X 0.0 Y 0.0
for Dreamer and Dreamer NX max Z size raised up to 150mm.

tckb, 2021-04-07:
I second this! if you remember in https://github.com/moonglow/FlashForge_Marlin/issues/9#issuecomment-813612576  I had the same issue, the bed origin I assumed was in center of the bed but with this firmware I had to adjust the center of bed to 0,0 in my slicer and in octoprint.

moonglow, 2021-04-07:
yep! i only now found a way how to do it better ( i found all offsets from original firmware ) all three models has different offsets and limits and now Marlin has same values 😎

tckb, 2021-04-07:
I see it as released in 0.7. I will test it

tckb, 2021-04-11:
Tested and works on dreamer nx

closed 2021-04-19 by moonglow