moonglow/FlashForge_Marlin #14: Nozzle Park Position on Dreamer Nx

tckb, 2021-04-08:
The Park position currently is placed in a weird place which isn't really convenient for changing filament.  Normally the park position is out of the printable area meant for purging filaments.

The current park position is 20,140,22
[image]
  
Park  position should be 270,150 

Park position is defined by  `NOZZLE_PARK_POINT` 

refer: https://marlinfw.org/docs/gcode/G027.html

Also, the save and go to park position is not enabled.  This needs `PARK_HEAD_ON_PAUSE` to be enabled - 

ref: https://marlinfw.org/docs/gcode/M125.html

moonglow, 2021-04-08:
>Park position should be 270,150

You mean something like right-front position ? center of bed is 0,0 and this position now out of border.  Yep i think it is reasonable.

tckb, 2021-04-08:
@moonglow I mean right-back position (close to the rear of the printer). with the new changes to the bed origin I am not too sure, the exact position but this is the  x-soft-limit, y-soft-limit . I don't really care of the z, you could raise the z to 2 mm. 

I want the back because there's already a place for filament dump (the space for 'right' spool holder, in nx this is empty)

moonglow, 2021-04-08:
> the space for 'right' spool holder, in nx this is empty

Ok i will try to find better place for Dreamer and Inventor too

tckb, 2021-04-10:
@moonglow any plans to release it in v0.8?

moonglow, 2021-04-10:
if new parking position is OK for you ( latest firmware from #12  ) i will release it in few days, need to make some other improvements.

tckb, 2021-04-10:
whats the current park position in #12  I'd prefer the x and y soft-limits. I don't mind waiting couple of days ;)  setting my filament routine

moonglow, 2021-04-10:
> whats the current park position in #12

 `X_MAX - 10, Y_MIN + 10` as universal solution ( RIGHT FRONT ) for all ,but for you i will build special version with  *_la and   `X_MAX - 10, Y_MAX - 10 ` if you want. Right-front looks comfortable for me you can easy access to nozzle to clean and so on

tckb, 2021-04-10:
thanks @moonglow  I will test both the solutions to see which one works.

tckb, 2021-04-11:
After testing v0.7_la 

I would like a universal solution rather having a tailored solution, although I'd recommend `X_MAX, Y_MAX - 10`  the 10mm is a bit close to the bed.

tckb, 2021-04-19:
closing this because i configured it in the fork

closed 2021-04-19 by tckb