moonglow/FlashForge_Marlin #17: False positive in Thermalrunway Protection? in v0.7

tckb, 2021-04-11:
As I understand Marlin 2.x already offers thermalrunaway protection but, with v0.7 I get this alert from octoprint that is this disabled.  This could be a false positive but, nevertheless should be addressed. I do not want to ignore this message. 

[image]

Can you verify if everything is in order? especially since the code is not available to check.

moonglow, 2021-04-11:
Our firmware has defined such values:
```
THERMAL_PROTECTION_HOTENDS 
THERMAL_PROTECTION_BED     
```
So... thermal runaway protection is ON, i think it is false positive warning,  you can read more information about it here ( offcicial forum ) [LINK](https://community.octoprint.org/t/octoprint-tells-me-that-my-printers-firmware-lacks-mandatory-safety-features-what-does-this-mean/350/1) . Octoptint use very tricky way to check it :) AUTHORS  string parsing is also fun...

> especially since the code is not available to check.

;)

 M115 says:
```
CAPABILITY: THERMAL_PROTECTION: 0
```
i will try to check sources maybe i disables chamber thermal protection ( actually we do not have heater for chamber, but it need for chamber rear fan control ) you can ignore it for now without any problem this is not a problem

tckb, 2021-04-11:
>you can read more information about it here ( offcicial forum ) LINK . 

yes I did read the entire thread, and there isn't a way to do a 'real' check without simulating an actual thermal runway check. 

>  ;)

good time to release the source :P I think its been stable enough. I am willing to test for a couple of releases before I switch back ;) 

>M115 says:
CAPABILITY: THERMAL_PROTECTION: 0

Ah! may be that's why?

moonglow, 2021-04-11:
> good time to release the source :P

i need to improve some code before publish it, there no errors here, just Octoprint behavior, be patient ;)

closed 2021-04-12 by moonglow

tckb, 2021-04-12:
>i need to improve some code before publish it,

what are the improvements you are planning?

moonglow, 2021-04-12:
i want to add support for Toshiba FlashAIR SD cards