moonglow/FlashForge_Marlin #22: Marlin 2.0.8 beta test for FlashForge Dreamer NX/Dreamer/Inventor

moonglow, 2021-05-05:
Marlin introduce new 2.0.8 version with lots of improvements and other, read more here: [Marlin 2.0.8](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.8)

- [classic_ui_05052021.zip](https://github.com/moonglow/FlashForge_Marlin/files/6426561/classic_ui_05052021.zip) archive - binaries with Marlin modern multi-language ui ( English/Finnish )
- [color_ui_05052021.zip](https://github.com/moonglow/FlashForge_Marlin/files/6426562/color_ui_05052021.zip) archive for old school guys - binaries with Marlin classic multi-language ui ( Russian/English/Finnish )

p.s: this is experimental build make sure what you understand it before install

moonglow, 2021-05-06:
So... after few hours of printing i can say what it works well stable for me ! I think now i can recommends this build for testing to everyone

Lucien-X, 2021-05-06:
@moonglow 
Seems great!
But now I am stuck with the v0.9.1 firmware published yesterday.
https://github.com/moonglow/FlashForge_Marlin/issues/24
Any help?

moonglow, 2021-05-21:
synced with upstream repo

moonglow, 2021-05-31:
update to 2.0.8.2, add MKS UI support

closed 2021-05-31 by moonglow