moonglow/FlashForge_Marlin #23: Dreamer NX no extrution with cura via usb

Lucien-X, 2021-05-05:
First of all, This project is AMAZING. 
Give us an alternative choice to have full control of our printer.

Meanwhile i run into some trouble when i print my test run.
I am using v0.9.1 firmware.
when i print things with cura via usb,
there seems to be no extrution while the extruder moving around.
I can hear extruder making a lot of noise, thus no filament coming out of it.
[image]

Thats strange, because when i use the original FF firmware,
I slice in Cura and send .g files to printer with FlashPrint, everything is predictable.

Wondering which step is wrong during my configuration.
Eager for some help

tckb, 2021-05-06:
@Lucien-X  may be its  basic, is the bed properly leveled? Iv'e had this problem numerous times 99% of time the bed was too close to the hotend. I'd say try  to re-level it as the first step

moonglow, 2021-05-06:
Hi @Lucien-X  ! Thank you! )

Original firmware use slightly different G-Codes, you need to change your `Start G-Code` and `End G-Code`. This profile is used only with original firmware.

Go to `Settings->Printer->Manage Printers...` select your printer and push `Machine Settings`

Locate you start and g-codes window, as example you **original** start G-Code:

```
;Start Gcode
G90 ;absolute positioning
M118 X25.00 Y25.00 Z20.00 T0
M140 S{material_bed_temperature_layer_0} T0 ;Heat bed up to first layer temperature
M104 S{material_print_temperature_layer_0} T0 ;Set nozzle temperature to first layer temperature
M107 ;start with the fan off
G90
G28
M132 X Y Z A B
G1 Z50.000 F420
G161 X Y F3300
M7 T0
M6 T0
M651
M907 X100 Y100 Z40 A100 B20 ;Digital potentiometer value
M108 T0
;Purge line
G1 X-110.00 Y-60.00 F4800
G1 Z{layer_height_0} F420
G1 X-110.00 Y60.00 E17,4 F1200
;Purge line end
```
M140 - Set a new target temperature for the heated bed and continue **without waiting**, [Marlin M140 description](https://marlinfw.org/docs/gcode/M140.html)
M104 - Set a new target hot end temperature and continue **without waiting**. , [Marlin M104 description](https://marlinfw.org/docs/gcode/M104.html)

Because Marlin has cold extruder protection **you will not get extrusion**.

```
M907 X100 Y100 Z40 A100 B20 ;Digital potentiometer value
```
This is most **dangerous** part :) i accept motor current as current value as example 0.7 for 0.7 Amps and so on ( it is better to read by eye ) ,but original firmware use raw values for digital potentiometer and this string for Marlin firmware will rise your motor current to **MAXIMUM POSSIBLE VALUES**. 

So, your new start G-code will looks something like this

```
;Start Gcode
G90
M140 S{material_bed_temperature_layer_0}     ;Heat bed up to first layer temperature
M190 S{material_bed_temperature_layer_0}     ;Wait for Heat Bed temperature
M104 S{material_print_temperature_layer_0}    ;Set nozzle temperature to first layer temperature
M109 S{material_print_temperature_layer_0}    ;Wait for Extruder temperature
M107 ;start with the fan off
G28
M117 Purge extruder
G1 Z50.000 F420
;Purge line
G1 X-110.00 Y-60.00 F4800
G1 Z{layer_height_0} F420
G92 E0
G1 X-110.00 Y60.00 E17,4 F1200
;Purge line end
G92 E0
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
```

And your end G-Code will looks like this
```
;end gcode
M140 S0 ; turn off heatbed
M104 S0 ; turn off temperature
M107 ; turn off fan
G28 X Y
M84 X Y E ; disable motors
```

So, main problem is a cold extrusion and another one is a different G-codes meaning

tckb, 2021-05-06:
Ah yes!  the profile Egon has prepared is for flashforge firmware ((https://github.com/KeltE/Flasforge_DreamerNX_Cura_Definition/blob/9fafb45302cc8b0687357044af564f246e2da2b9/Flasforge_DreamerNX_Cura_Definition/resources/definitions/flashforge_dreamer_nx.def.json#L34) )  so yes the as @moonglow  mentioned the start and end-gcode for that profile  doesn't work. 

I've had the same issues at the beginning : https://github.com/moonglow/FlashForge_Marlin/issues/9

Lucien-X, 2021-05-06:
@moonglow 
You are absolutely right about that motor current thing.
The profile provided by @KeltE is pretty nice, **but not compatible with the marlin firmware** .

Those extra Amps makes the extruder gear jiggling a lot and pretty noisy,
the filament got chewed up in one spot by the feed gear, and then failed to feed any further. 

I reconfigured Cura with the sample g-code written by @tckb , 
then reload the filament and releveled the bed.
Then print via usb again, and it works like charm ~

BTW, I think you should append those sample g-codes to README or the installation manual ,
to avoid further damage to the printers of the late-comers.

Lucien-X, 2021-05-06:
One more thing, I am using Windows10 OS.
I can't update the firmware using the firmware tool provided,
looks like the later FF DNX firmware is using VID 0x2B71 instead of  0x0315 (You can see that from device manager in win10)

So I managed to use the official FlashPrint firmware update function to do the samething.

Here is my procedure:

Go to X:\Program Files\FlashForge\FlashPrint\firmware\FlashForge Dreamer NX  (X is your installation disk location)
and copy the firmware files to that folder like this (maybe the intro file is not necessary, I didn't check)
[image]

Open `FlashPrint => Tools => Update Firmware`
and do the things below
[image]

reboot the printer after the software is done, and the firmware is marlin now

closed 2021-05-06 by Lucien-X

icebluscorpion, 2025-03-28:
i have also no extrusion but on my Dreamer the extusion is reversed when it executes a print. but the load filament routine works fine. i have also tried the version with _dir_ in it but sill does the revers extruding.