moonglow/FlashForge_Marlin #24: Firmware tool not working on Dreamer NX later model

Lucien-X, 2021-05-06:
I am using Windows10 OS.
I can't update the firmware using the firmware tool provided,
looks like the later FF DNX firmware is using VID 0x2B71 instead of  0x0315 (You can see that from device manager in win10)

So I managed to use the official FlashPrint firmware update function to do the samething.

Here is my procedure:

Go to X:\Program Files\FlashForge\FlashPrint\firmware\FlashForge Dreamer NX  (X is your installation disk location)
and copy the firmware files to that folder like this (maybe the intro file is not necessary, I didn't check)
[image]

Open `FlashPrint => Tools => Update Firmware`
and do the things below
[image]

reboot the printer after the software is done, and the firmware is marlin now.

To get back to original firmware, I asked for the 
[Dreamer NX TF Card Files Backup.zip](https://github.com/moonglow/FlashForge_Marlin/files/6435575/Dreamer.NX.TF.Card.Files.Backup.zip)
from FF Customer Support Agency.

Haven't try that to recover the firmware yet.
Maybe I'll test that later

Lucien-X, 2021-05-06:
After what I've done, it is not possible to recover with the same way.
So sad I can't update firmware with ff_flasher either. 😂 LOL

Seems that my PC can't recognize the printer as a DreamerNX USB Device, 
but treat it as a USB Serial Device on COM3.
[image]

Anybody any help PLZ?

tckb, 2021-05-06:
@Lucien-X  :P Yes! once you are in Marlin, Flashprint doesn't recognize your printer. I don't have a windows to test but, if you are able to run Bash via the LinuxSubsystem  in Windows, I can adapt the `install` on my fork to may be reflash FYI - https://github.com/tckb/FlashForge_Marlin/wiki#restoring-to-stock-firmware

moonglow, 2021-05-06:
@Lucien-X  you can use  Linux script or Windows utility to update firmware, but you can also use FlashPrint for that, FlashPrint will not works from firmware mode ( works from firmware mode if you runed on from original firmware ), you must catch a **bootloader** via python script or via ff_tool or via **FlashPrint** tool also ( yep you can use FlashPrint under Mac, Linux or Windows for that too ). After printer starts and logo was appear on screen it will recognized with different PID/VID ( you can see that in your windows device manager or dmesg output under Linux ) script or utility will wait for it and try to communicate with bootloader, whats mean what you must run it before printer be turn on. But while you use script or utility your FlashPrint tool must not be runed.

p.s: for **ff_flasher** try to put firmware to the **same** folder what ff_flasher located

tckb, 2021-05-07:
@Lucien-X  perfect, due to lack of proper testers, I do not have much of tooling around windows. you could either use linux (if it allows you to access the usb) or macOS, here's a good starting point: https://github.com/tckb/FlashForge_Marlin/wiki#flashing-per-compiled-release-versions

Lucien-X, 2021-05-07:
@tckb 
According to 
Jetguy from FlashForge google group:
https://groups.google.com/g/flashforge/c/E3ce2VTVQJk/m/SPzV7-EzAwAJ
```
Directly access the bootloader has to be done on a Windows 7 machine. 
Windows 8 and higher likely will not work ,
as the driver signing is likely an issue ,
and you won't be able to connect and recover ,
because the driver does not load. 
```
So I managed to get things done with a "plan B", the virtual machine mechanisum.
I use [VMware Workstation 16](https://download3.vmware.com/software/wkst/file/VMware-workstation-full-16.1.1-17801498.exe) and have my ubuntu virtual machine installed  (**Notice that VMW 15 is not compatible with Win10 either, that would crash your system when you attempt to access any USB device**) , then followed the [Linux-instalation-guide](https://github.com/moonglow/FlashForge_Marlin/wiki/Linux-instalation-guide)

@moonglow 
There is a small flaw within that guide, instead of
`ff_flash_firmware.py any_dnx_firmware.bin`
you should prepend 'sudo':
`sudo ./ff_flash_firmware.py ./any_dnx_firmware.bin`
or you may encouter some error during firmware upload.

Lucien-X, 2021-05-07:
> First of all it is very nice what you solved the problem, this is most important part ! Jetguy told about old version of FlashPrint in 2016, latest official version FlashPrint x64 works fine and all drivers installes OK ( i tested ff_flash tool with Windows 10 x64 and it works for me ). About **sudo**, i think you forget to install FlashPrint for Linux ( installer will place udev rules to access USB device without **sudo** ).

Yeah, I forgot to install the Linux version of FlashPrint. That must be the reason of sudo requirement.
And maybe there is something wrong with my Win10 x64 environment, and that is really hard to find out why ,
hope the late-comers won't run into same issue as I do.

And at the end of the day , 
Virtual machine rules ! φ(゜▽゜*)♪  Windows Sub Linux sucks ! (╯▔皿▔)╯

closed 2021-05-07 by Lucien-X

tckb, 2021-05-07:
Good to know that you fixed it @Lucien-X  yeah dealing with windows is always as crap as it advertises