moonglow/FlashForge_Marlin #25: DreamerNX chamber fan not working properly in v0.9.2

Lucien-X, 2021-05-09:
Just flashed 0.9.2 firmware , the cooling fans on the back of chamber is not working right.

There are two of them , but when I  adjust 'Fans Speed 3' manually , only one of them is spining as I wish.

moonglow, 2021-05-09:
This fans connected to ONE output it not controlled separately. To make sure what you disable automatic chamber fan control, please set chamber temp to 0 ( it sets to 45 by default ) and then try to test FAN3 control. I recommend to set it to 100% and if your second fan is stuck, try to turn it by hand and this try again. I do not know why one of them is works such they, because where is not other output. Maybe one of fan does not like PWM or something like that.

Lucien-X, 2021-05-10:
False alert, the same shit happens when I roll back to original firmware.
Gotta be the hardware problem. I'll enquiry the seller about this.

closed 2021-05-10 by Lucien-X

Lucien-X, 2021-05-10:
Nailed it, the PWM signal is intact , proved that with a n alternative fan from my PC.
Gotta order a new 8015 fan to replace that broken one.

[image]
[image]

[image]

moonglow, 2021-05-10:
@Lucien-X RGB fan looks much better )