moonglow/FlashForge_Marlin #26: MKS UI for FlashForge ?

moonglow, 2021-05-13:
I think it looks pretty and more user friendly ( it also has builtin multi language support and model preview mode and UI skins can be customized by user without firmware compilation ), but it needs to add an external cheap SPI FLASH module ( it can be easily connected to WIFI socket ). New FlashForge also can handle MKS WIFI module ( it based on cheapest possible WIFI module EPS8266 ). Motherboard WIFI socket pinout: [LINK](https://github.com/moonglow/FlashForge_Marlin/wiki/FlashForge-WIFI-module-pinout). I'm waiting for modules, to do some tests ( now it runned with emulated flash memory on SD card ,but it works verrry slowly )

----
Some screenshots of MKS UI:
[image]
[image]

tckb, 2021-05-13:
I really like the MKSUI but problem is this custom mods that has to be fixed in.

moonglow, 2021-05-13:
Hm... can you provide me more information ?

tckb, 2021-05-14:
@moonglow  I haven't upgraded to 2.0.8 (lack to time) but as you mentioned in the description, it needs extra components to work with Stock FF board isn't it?

moonglow, 2021-05-14:
@tckb yep ,but this is easy to install solution you no need to solder it to motherboard just connect it with DuPont wires and that's all folks :)  you can switch back easily all extension boards connected to WIFI socket. This is optional solution, user can choose CLASSIC_UI MARLIN_UI or MKS_UI. As example we can install TMC2209 stepper motor drivers to our original board to make it silent and get better printing quality, this is also changes to original board, but it also optional solution.

[image]

closed 2021-05-24 by moonglow