moonglow/FlashForge_Marlin #30: Detailed installation for mks firmware.

andreacampanella, 2021-06-08:
I just tried the MKS firmware and it took me a while to get that I need an external eeprom to run it, perhaps that should be a warning somewhere...

moonglow, 2021-06-08:
Hi @andreacampanella ! I added WIKI section for it [MKS UI installation guide](https://github.com/moonglow/FlashForge_Marlin/wiki/MKS-UI-based-firmware-installation-guide), if you have any question i will new info to that guide, but i recommends you to use Marlin Color UI, it pretty nice looking UI and well tested ( my main setup ). MKS UI is really cool and international and i added it to make use a choice to any of them.

andreacampanella, 2021-06-08:
That's awesome, thanks.

I like the color UI but the touch interface it's a bit scary, one wrong touch and you click the wrong menu, I prefer the classic.