moonglow/FlashForge_Marlin #31: Instructions for Dreamer.

andreacampanella, 2021-06-08:
I had to remove the wifi card to get the sdcard working, this is not mentioned anywhere.

moonglow, 2021-06-08:
You no need to remove Wi-Fi card to get SDCARD working, hm... i never heard about it from any other Dreamer users, need more info. I will ask other users about it too.

andreacampanella, 2021-06-08:
with the wifi card the SDCARD wouldn't init correctly, in the original firmware you can't use wifi and SDCARD at the same time, I suspect they share the same SPI port, I removed the wifi card and it started to work.

Also, do you know why there is an sdcard inside as well?

moonglow, 2021-06-08:
>suspect they share the same SPI port

Yep, but they use different chip select gpio so where is no problem with same SPI port, as an example if you use MKS UI you will share SPI between SDCARD and SPI FLASH memory and it works well. Original firmware does not use it at same time maybe because it is internal architecture limits, original firmware has many strange things... This is why i use Marlin firmware for my printer. I will recheck this issue with classic and color firmware.

>Also, do you know why there is an sdcard inside as well?

Internal SDCARD connected to SDIO interface and used as main storage to save data via WiFi module, holds bootloader main LOGO and firmware before upte and so on. I do not want to use it because my main goal do not modify internal structure of printer what can be difficult to fix ( to remove internal sdcard you need to disassemble printer ), that's why i also do not use internal board EEPROM. And you can easily revert back to stock firmware without any cons.

andreacampanella, 2021-06-08:
Do you use Prusa slicer by the way? would you share the profile for it?

moonglow, 2021-06-08:
@andreacampanella looks like i know why do you have a WiFi issue... before i introduced MKS UI support i never turn on WiFi socket module power and it be off by default, can you check this build with WiFi module installed. I think it will fix your problem, if so i will push changes and make new builds.

[fw_2.0.8.2_dreamer_06082021.zip](https://github.com/moonglow/FlashForge_Marlin/files/6618168/fw_2.0.8.2_dreamer_06082021.zip)

>Do you use Prusa slicer by the way? would you share the profile for it?

I have a one head model Dreamer NX, but i do not make any special thing, but anyway here is it:
 
[config.zip](https://github.com/moonglow/FlashForge_Marlin/files/6618189/config.zip)

moonglow, 2021-06-14:
fixed here [commit](https://github.com/moonglow/FlashForge_Marlin/commit/106ea25d3084d47d3a6d7681f4c4a7c40f8e58ea), you can use regular builds now

closed 2021-06-16 by moonglow