moonglow/FlashForge_Marlin #34: Bed Leveling 3 Points on Dremel 3D20

OneMonkeyArmy, 2021-06-19:
The 3 point leveling works exactly like before, except that after the last point, it stays at the same place. And if the nozzle is hot, it burns the the adhesive sheet on my build plate. Actually, it makes a bump or a crater, depending on how fast i move the z axis. You shoud, or I should add just one line to send the nozzle to a parking spot , away from the buildplate ! or even better, the homing position, back left corner, of course after lowering the buildplate of a fixed amount. (Z safe distance). I never slept since yesterday, I have tried to compile the code from you, but always have problems with include path...  I have been able to print a Benchy (the small boat) to a relative good quality in less than 10 minutes !

moonglow, 2021-06-19:
Hi Daniel!

3 point leveling added through Marlin `User Menu` just as G-code script and it can be changed to be more useful, script itself:

```
G90
M117 Bed leveling started
G28
G1 X-5.3 Y-71.1 Z10 F3000
G1 Z0 F800
M0 Adjust the screw and press to continue
G1 X-54.4 Y61.6 Z10 F3000
G1 Z0 F800
M0 Adjust the screw and press to continue
G1 X42.8 Y61.6 Z10 F3000
G1 Z0 F800
M0 Adjust the screw and press to continue
G1 X0 Y0 Z10 F3000
G1 Z0 F800
M117 Leveling done
M84
```

I totally agree with you, that stay at last point is not a good idea. If you provide better version i will accept it or i can change it soon with take into account your suggestions.

moonglow, 2021-06-21:
Bed leveling script changed ( do a little Z rise at end got to X Y home position ):
```
G90
M117 Bed leveling started
G28
G1 X-5.3 Y-71.1 Z10 F3000
G1 Z0 F800
M0 Adjust the screw and press to continue
G1 X-54.4 Y61.6 Z10 F3000
G1 Z0 F800
M0 Adjust the screw and press to continue
G1 X42.8 Y61.6 Z10 F3000
G1 Z0 F800
M0 Adjust the screw and press to continue
G1 X0 Y0 Z10 F3000
G1 Z0 F800
M117 Leveling done
G1 Z5 F420
G28 X Y
M84
```

OneMonkeyArmy, 2021-06-23:
There is still an error in your script, but i will send you the corrected one! one important line is missing!
It needs a M0 Check center alingment and press to continue....  before M117 Leveling done, to let user check the final alignment in the center of the bed.

moonglow, 2021-06-23:
Ahh... yep :) thank you ! i will fix it for next build !

moonglow, 2021-07-01:
Fixed

OneMonkeyArmy, 2021-07-06:
yep!

OneMonkeyArmy, 2022-01-09:
Just an idea,, did

moonglow, 2022-01-09:
>when compiling, will solve the problem for all users...

Such things already used for other printers, read more about Dremel here #67 .If someone provide me pretested script for Dremel 3D20 i will integrate it, but nothing was provided.

moonglow, 2022-01-19:
Closed for no activity, use #67

closed 2022-01-19 by moonglow