moonglow/FlashForge_Marlin #35: Dreamer: Can not update Marlin

me-no-dev, 2021-06-21:
Hi :)

I have FF Dreamer and I flashed it with the latest prebuilt `fw_2.0.9_dreamer_swap_la_06162021`.
I found issues with the `swap` config (MIN/MAX X) so in order to fix them, I manually compiled new firmware.
I tried uploading it with `sudo ./ff_flash_firmware.py fw_2.0.9_dreamer_swap_la_06212021.bin` but regardless what I tried, it will timeout reading from the USB or will not write because the port went away. I do not see it able to stop the boot process at all and bootloader times out going back into the firmware.

Given that I can upload the firmware manually to the internal SD, is there a way to trigger the upload process?

moonglow, 2021-06-21:
Hi !

> I tried, it will timeout reading from the USB or will not write because the port went away

If you get timeout issue you can rerun script immediately to retry flash attempt or use FlashPrint or windows flash utility as alternative way ( they also works with bootloader ), but you can also use manual update procedure via invoking bootloader update trigger. Copy ( i assume your printer is FF Dreamer ( not NX or Invenor ) ) firmware with name `dreamer.bin` to `sys` folder of internal **uSD** card ( that card located on motherboard and yep you need to remove bottom shield to access it ), next you need to edit update script and comment or remove lines from #78 to #99. 

p.s: maybe i need leave update trigger menu for all printers not only for Dremel as alternative way in such situations :roll_eyes:

me-no-dev, 2021-06-21:
Thanks for the fast reply @moonglow :)

Printer bottom is open. the new `dreamer.bin` is in the sys folder.

Editing the flash script will not help, because it can not connect to the printer reliably. FlashPrint is also failing and since I am on a Mac, I can not try the exe :(

I read through the changes for the update trigger and what I gather is that it wipes particular sector in the flash and writes back the printer serial to it. I am not very familiar with STM32 and SWD, but I do have an adapter and can probably connect to the printer and wipe the flash manually?

me-no-dev, 2021-06-21:
Tried the Windows app and result is the same. I guess my bootloader is not OK and is not communicating back with the host. I need some manual way to trigger the update

moonglow, 2021-06-21:
>FlashPrint is also failing and since I am on a Mac, I can not try the exe :(

I'm lack of Mac so i think i can't reproduce it to understand what can be wrong, but looks like you updated it already via script to get Marlin working ? Most fast way to fix that problem it is to find any Linux or Windows system

>Tried the Windows app and result is the same. I guess my bootloader is not OK and is not communicating back with the host. I need some manual way to trigger the update

Hm... bootloader must works in any case only old bootloaders from Dremel machine does not allows users to update firmware via bootloader

>I read through the changes for the update trigger and what I gather is that it wipes particular sector in the flash and writes back the printer serial to it.

Yep bootloader has a special magic string ( encrypted MCU serial + assci string ) it used to protect bootloader from running on different MCU ( MCU serial is hardware ID and unique for each of them ). **First of all you to make full backup of flash to make sure that you can bring it to live from any state**. On our luck MCU is not locked :)

 >I am not very familiar with STM32 and SWD, but I do have an adapter and can probably connect to the printer and wipe the flash manually?

If your have a ST-Link or similar adapter you can do anything with that board, but for feature purpose is better to fix that issue, looks like libUSB works different on Mac :( But you can connect via SWD interface and flash **unencrypted** firmware to their base address: `0x08010000`

[image]

DEBUG socket it is 3 pin SWD interface, pinout ( pins numbering from picture ), you can use dupont wires for that:
1 - **SWCLK** - CLK ( pin 109,  PA14 )
2 - **SWDIO** - DIO ( pin 105, PA13 )
3 - **GND**

If your adapter needs **VTarget** voltage you can get it from **VCC 3.3V pin**

 p.s: make sure you set config to your jtag adapter to using **SWD** interface and select correct MCU **STM32F407ZG**

me-no-dev, 2021-06-21:
> Hm... bootloader must works in any case only old bootloaders from Dremel machine does not allows users to update firmware via bootloader

My machine predates any work with Dremel (August 2014). So it can have anything on it.
I already have things like dfu-util and openocd on my computer. Do you have the command I need to execute to dump and write the flash back? I guess by "unencrypted" you mean the firmware bin before going through the flashforge_firmware_tool?

moonglow, 2021-06-21:
>My machine predates any work with Dremel (August 2014). So it can have anything on it.

Very interesting... But how you did the first Marlin update ? Through application mode ? I think after you fix your problem you can flash updated bootloader to easy update in fiature i will upload it soon.

>I already have things like dfu-util and openocd on my computer

What JTAG adapter you have ? If you use ST-Link, you can use CubeMX programmer ( version for Mac is also available ): https://www.st.com/en/development-tools/stm32cubeprog.html

>I guess by "unencrypted" you mean the firmware bin before going through the flashforge_firmware_tool?

Yep, before it was encrypted by flashforge_firmware_tool or you can decrypt stock firmware with same tool also.

me-no-dev, 2021-06-21:
here is the current state. start at 0x08000000 with size of 1M

[broken-bootloader.bin.zip](https://github.com/moonglow/FlashForge_Marlin/files/6687112/broken-bootloader.bin.zip)

moonglow, 2021-06-21:
@me-no-dev !!! Thank you so much for this bootloader version yep this is old version.
Try to flash this dump ( it has updated bootloader v1.4 ) + original firmware v2.15.20200917 if all be Ok i think you can now use classic way to install Marlin or revert back to stock firmware. 

[fixed_dump.zip](https://github.com/moonglow/FlashForge_Marlin/files/6687278/fixed_dump.zip)

me-no-dev, 2021-06-21:
Thanks a bunch @moonglow :) Will try this now and will also enable the Update trigger to make sure that all options work :)

closed 2021-06-21 by me-no-dev