moonglow/FlashForge_Marlin #41: Need  touch screen calibration option !

OneMonkeyArmy, 2021-07-12:
My touch screen on my dremel 3d20 is not perfectly aligned, when I was running the old firmware from Dremel, there was a menu to calibrate the touchscreen, by touching each corner, and the center... is it possible to add easily?

OneMonkeyArmy, 2021-07-12:
already added, forget about it !

closed 2021-07-12 by OneMonkeyArmy

moonglow, 2021-07-13:
yep you can press on empty space on main screen or invoke it via `M995`

OneMonkeyArmy, 2021-07-13:
... or directly... ui.goto_screen(touch_screen_calibration) 

Anyway... I already spent too much time trying to improve that GUI, that is very ugly and not user friendly...

And it is a mix of text menu item  (old marlin) and Touch with GUI, more recent... 

I understand 90% of it, but I wonder why the guys didn't add, at least a few comments !!!

Anyway, my fingers are too big for this king of GUI. And it is too much work to get a nice one with Icons.

I think I will give up on that thing...

If it was just for me, I would just changed everything, controller+lcd, but my goal was to add the electronics (that is working) and have a nice GUI, so many people will be interested.. but like that ... I'm pretty sure people will just skip!! 

I will go back to my MONSTER system...