moonglow/FlashForge_Marlin #46: Reverse Polarity of E2 Extruder?

DDMH, 2021-08-01:
I have flashed "fw_2.0.9.1_dreamer_swap_06302021.bin" to my Dreamer. When I use "Change Filament"->"Load E2" (right extruder), the right extruder seems to pull the filament out instead of pushing it in. The Dreamer dual extruder assembly is similar to this one in the image below:

[image]

Since the two extruder motors rotate in opposite directions when printing, I think the polarity of E2 needs to be flipped in the firmware.

moonglow, 2021-08-05:
Hm... very strange thing looks like they must goes in to the same direction and i never saw such things before.Thank you i will build special version for you and if all be OK we need to think how it can be resolved while autobuild.

DDMH, 2021-08-05:
Today I tested it again and the motor of E2 was in reverse. When I used "Unload E2", the filament was actually being loaded. Same thing happened when I sent Gcode directly through Pronterface.

[image]

moonglow, 2021-08-05:
If your stock extruder do same and original FF firmware works OK with that, you can try this version:

[e1_dir_inverse.zip](https://github.com/moonglow/FlashForge_Marlin/files/6941304/e1_dir_inverse.zip)

DDMH, 2021-08-05:
Thank you, moonglow! I will test it later today and report back.

The first photo (Flexion extruder) was only used to show the dual extruder assembly, because it's similar to the stock one. I still have the stock extruders on my Dreamer. Sorry for the confusion.

DDMH, 2021-08-06:
@moonglow I tested using the "Load E2" option and extruding manually by sending Gcode via Pronterface. Both methods worked flawlessly. Thanks again for the quick patch.

closed 2021-08-06 by DDMH