moonglow/FlashForge_Marlin #48: [POLL] Which Marlin UI you prefer ?

moonglow, 2021-08-10:
/polls Color UI 'Classic UI' "MKS UI"

OneMonkeyArmy, 2022-02-16:
Good news for everybody, and bad news for me... I have decided to slowly, a step at a time, re-implementing my GUI as an independent module, that will be easy to update with the newer version of Marlin. I will create a fork, and add all the needed graphics, files, library and custom code... etc
But, the first step will be to update the LCD driver and graphics class, as a different library, because, it is impossible to use the actual ones, as the are. And modifying all the files ( as the actual method used by my GUI, is a complete mess, when it comes to update Marlin) I will do it only if some people get involved in the project, because alone, it is a very long and lonely project !) 

AND yes, yes, yes, You were right, Moonglow, It was a bad Idea to do it with a one man team ! even if my nickname is the One Monkey Army...

moonglow, 2022-02-16:
Sorry ,but i can't accept your invite, because i haven't so much free time for my projects to helps with some side project, but you goes to open source and that great news. I hope your project will be successfully implemented !

closed 2022-08-14 by moonglow