moonglow/FlashForge_Marlin #51: FF dreamer T1 unloading instead of extruding

exxosuk, 2021-08-13:
Having given up with my dreamer firmware I flashed the latest Marlin software. Had a lot of fun with the G-codes not being compatible. The left head is printing perfectly fine, however the right head still of extruding, is retracting filament ?!

I am using pronterface to set the head temperature and the tool, and when I clicked extrude, the left head extrudes filament as expected, changing tool to the right head, it retracts it. So no idea what is going on with that ?

moonglow, 2021-08-13:
@exxosuk  Hi!

Please read this first: #46

Looks like Dreamer old motherboard really has different directions for left and right extruders, i will fix it ( new build variant ) soon. You can use firmware what you can download from previous discussion for now as temporary solution.

>Had a lot of fun with the G-codes not being compatible.

Yeh.. but Marlin if more popular standard for 3D printers instead of Sailfish based firmware.

exxosuk, 2021-08-13:
Thanks. I had not seen that post :)  i'm trying to calibrate with the left head, so plenty to do yet ;)

moonglow, 2021-08-15:
@exxosuk New [updates](https://github.com/moonglow/FlashForge_Marlin/releases/tag/0.12.3) now includes builds with swaped extruder direction for Dreamer old mother boards for all who can't build firmware yourself, you can try it from here, if i did all right it will swap only needed extruder :)

exxosuk, 2021-08-15:
@moonglow thanks so much for this!  both working in the correct direction now!

I did try compiling the firmware myself a few days ago, but it comes up a few errors and no idea why :(

moonglow, 2021-08-15:
Marlin has a good guide how to prepare toolchain setup and build it all ( imho if you use linux enveloperment it much more easy to do )

closed 2021-08-15 by exxosuk

exxosuk, 2021-09-07:
I'm still trying to figure out how to compile this myself. I've selected AT2560 from the menu. 
Uncommented  #define FF_DREAMER_MACHINE

But chucks out errors and no idea what its actually complaining about ?

`In file included from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\inc\MarlinConfigPre.h:52:0,
                 from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\inc\MarlinConfig.h:28,
                 from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\HAL\STM32\hotend\ads1118.cpp:1:
Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\HAL\AVR\inc\Conditionals_LCD.h:25:4: error: #error "Sorry! TFT displays are not available for HAL/AVR."
   #error "Sorry! TFT displays are not available for HAL/AVR."
    ^~~~~
In file included from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\pins\stm32f4\pins_FF_MOTHERBOARD.h:4:0,
                 from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\pins\pins.h:587,
                 from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\inc\MarlinConfig.h:34,
                 from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\HAL\STM32\hotend\ads1118.cpp:1:
Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\pins\stm32f4\env_validate.h:25:4: error: #error "Oops! Select an STM32F4 board in 'Tools > Board.'"
   #error "Oops! Select an STM32F4 board in 'Tools > Board.'"
    ^~~~~
In file included from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\inc\MarlinConfig.h:49:0,
                 from Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\HAL\STM32\hotend\ads1118.cpp:1:
Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\inc\SanityCheck.h:1889:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
        ^~~~~
Z:\FlashForge_Marlin-0.12.3\FlashForge_Marlin-0.12.3\Marlin\Marlin\src\HAL\STM32\hotend\ads1118.cpp:9:10: fatal error: pinconfig.h: No such file or directory
 #include "pinconfig.h"
          ^~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
`

moonglow, 2021-09-09:
@exxosuk 
Our printers is not based on AT2560, default envelope settings is already configured to build correct firmware. I think you need to read original Marlin manual how to configure build toolchains and so on. I recommends you to use Linux + VSCode + PlatformIO plugin if you want to build it yourself. 

p.s: please use discussion tab for such questions