moonglow/FlashForge_Marlin #56: Support for Monoprice MP Inventor I

wastelandgunner, 2021-09-05:
I recently acquired a Monoprice MP Inventor I. Is there any chance support can be added for it? I absolutely hate using Flashprint-MP. I also don't use Github often, so apologies if this isn't labeled correctly.

Thanks!

moonglow, 2021-09-09:
@wastelandgunner Hi ! Looks like it same base as original FF, i think it possible, but i need more info about them and assistance if they firmware encryption key is different. I will give more info asap

moonglow, 2021-09-14:
@wastelandgunner I can't find original MP slicer software and firmware files, but if we read user manual i think it really same as original FF Inventor, you can try to flash firmware for inventor printer and check if it be OK

p.s: if you has it please upload it here

wgaylord, 2021-09-14:
I have heard people have uploaded the original FF Inventor firmware onto the Monoprice Inventor to be able to use the normal FF Slice instead of the MP one.   I think I still have the software and might be able to find the firmware. https://www.reddit.com/r/FlashForge/comments/fvoj7z/monoprice_flashforge_inventor_1_and_octoprint_and/fo5sdv6?utm_source=share&utm_medium=web2x&context=3

wastelandgunner, 2021-09-14:
> I have heard people have uploaded the original FF Inventor firmware onto the Monoprice Inventor to be able to use the normal FF Slice instead of the MP one. I think I might be able to find the firmware file somewhere on my computer. As for the slice it seems like its a rebranded FF slicer. https://www.reddit.com/r/FlashForge/comments/fvoj7z/monoprice_flashforge_inventor_1_and_octoprint_and/fo5sdv6?utm_source=share&utm_medium=web2x&context=3

I did attempt to do this as well before coming here. I couldn't get Flashprint to even recognize the firmware unfortunately.

wgaylord, 2021-09-14:
weird.  Yea, I haven't tried it because I didn't want to mess up my brother and I printer only a few weeks after we got it.   Also just wondering once you install this firmware (the Marlin one) how would you even set up a slicer? I haven't been able to use any slicer with my printer except the MP Flashprint.

moonglow, 2021-09-14:
You can try to install firmware via my updater tool under windows ( [Updater](https://github.com/moonglow/FlashForge_Marlin/wiki/Windows-instalation-guide) ) but on **your own risk**, just download prebuild binaries for Inventor and try it. If printers bootloader use same VID/PID as original FF Inventor you will able to do that. Be sure what you have original firmware ( it must be located in your MP Print installation folder if they use same data structure as it does in FlashPrint ) to revert your printer back to stock firmware. 

p.s: if someone really installs already original FF firmware on it all be ok ;)

wastelandgunner, 2021-09-14:
I did attempt that, but your tool did not recognize the printer.

moonglow, 2021-09-14:
@wastelandgunner Can you provide me original MP slicer software, i think it will gives me more answers... i can't find original software download links on over Internet =( ? And if you can also USB VID/PID values for normal mode ( while printer already started and ready ) and for bootloader mode ( while printer boots it can change USB  VID/PID value ). You can use Windows Device Manager or Linux dmesg output. I think they just changeg a little some internal configs to be unique

wastelandgunner, 2021-09-18:
> @wastelandgunner Can you provide me original MP slicer software, i think it will gives me more answers... i can't find original software download links on over Internet =( ? And if you can also USB [...]

Apologies for the delayed response. Here are the values for the bootloader:

PID: 0x00f8
VID: 0x2b71

Values for the normal operation:

PID: 0x00f8
VID: 0x2b71

Seems to me like it doesn't change VID/PID between the bootloader and normal operation, yet I did notice a clear disconnect of the device as it transitioned from the bootloader to normal operation. Here is also the link to Flashprint-MP:
https://downloads.monoprice.com/files/software/33820_Software_Win64_v3.23.2_181120.zip

Thank you for your assistance!

moonglow, 2021-09-21:
@wastelandgunner  Thanx.... I recommends you first to save full firmware to be sure what you can bring back your printer from absolutely any state if some will goes wrong( [firmware backup guide](https://github.com/moonglow/FlashForge_Marlin/wiki/Backup-printer-firmware) ) it will also be helfull for my bootloaders collection ^_^. Because i can't find original firmware to test upgrade process =( Originally they was located here:  
- http://www.ishare3d.com/3dapp/public/FlashPrint-MP/firmwareInfo  ( firmware updates, not works for me )
- http://www.ishare3d.com/3dapp/public/FlashPrint-MP/appInfo1 ( software updates, works fine )

Looks like Monoprice droped support for they printers if anyone can contact them to obtain firmware it be also helpful.

# **For extreme lovers:**
- Download and install MP-Print from link above
- Download that archive with Marlin firmware: [MP Inventor I.zip](https://github.com/moonglow/FlashForge_Marlin/files/7201073/MP.Inventor.I.zip)
- Unpack it to `x:\Program Files\FlashPrint-MP\firmware\`
- Your folder must looks like that:
[image]
- Update firmware from MP Print ( Tools->Update Firmware ):
[image]
--------------
# How to update Marlin from Marlin or revert back to MP firmware or to FF Inventor firmware ( untested )?
- Fallow that guide from Dremel: [LINK](https://github.com/moonglow/FlashForge_Marlin/wiki/Installing-Marlin-on-Dremel-3D20)
- Don't forget you need to put Marlin or original firmware with `mp.bin` ( not `dremel.bin` !!! ) name to `sys` folder on your INTERNAL micro sdcard.

wastelandgunner, 2021-09-21:
Thank you so much for the update! I've gone ahead and purchased a ST Link V2. Should arrive in 2 days. I'll extract the firmware then and get back to you!

moonglow, 2021-09-22:
@wastelandgunner It would be nice, it gives me ability to flash this dump to my motherboard and write normal guide and make that process safety. Also it will be usefully for comunity to make sure what they can bring printer to life from any state to life

wastelandgunner, 2021-09-23:
> @wastelandgunner It would be nice, it gives me ability to flash this dump to my motherboard and write normal guide and make that process safety. Also it will be usefully for comunity to make sure what they can bring printer to life from any state

Following your firmware backup guide, this is what I extracted from the printer. Not sure if it's correct, considering it's 1024kb, but this is what I pulled:

https://github.com/wastelandgunner/Monoprice-Inventor-I-Firmware

wastelandgunner, 2021-09-24:
After using it for a bit, here's what I noticed:

Working:
- [x] TFT Display (new colors)
- [x] Touch screen controller
- [x] Bed and nozzle temperatures
- [x] Chamber RGB light
- [x] Cooling fans, endstops, 
- [x] All steppers
- [x] SD card
- [x] Flashforge WiFi Module

Not working:
- [  ] USB (detected within Windows, but not within Cura or Octoprint)

Attached are photos of the printer interface.

[image]
[image]

moonglow, 2021-09-24:
@wastelandgunner You update firmware via FlashPrint from my guide  ???? looks like i put original FF Inventor firmware and not Marlin !!!!!! ....  You can now update firmware to marlin just rename it same way.... It was really my mistake, sorry o_O

About dump:
Thank you! Yep they use same encryption keys as original FF printers so i think it safe to update printer to Marlin, i will also do some tests today to check it, but if FF Inventor firmware works fine, Marlin must works fine too. I will release new versions today with decreased Z height ( it was increased to 155, but it be not OK without bed recolibration, so i will revert it to 150mm ).

wgaylord, 2021-09-24:
Got a question, once we setup a Flashforge/Monoprice  printer with this firmware how do we print to it? As I have had any luck printer to my printer's current firmware with anything but the Monoprice Flashprint slicer.  (And hand modified Cura G code.)

moonglow, 2021-09-24:
@wastelandgunner 

New FF Flasher will allow update firmware from Bootloader/Application mode: [ff_flasher_2021_09_24.zip](https://github.com/moonglow/FlashForge_Marlin/files/7227700/ff_flasher_2021_09_24.zip)

So, now you can choose firmware from Release page and install Marlin or FF Inventor firmware. I will update new Marlin version soon with Z=>150 for Inventor Machine, but you can try to load and check Marlin now ( do not forget what Z-MAX sets to 155 not to 150 for now )

wgaylord, 2021-09-24:
Okay, but like I never could get the start and end g-code figured out for my machine. (Part of why I never got Cura working with my current machine without hand modifying the start and end code.)

wastelandgunner, 2021-09-25:
> @wastelandgunner
> 
> New FF Flasher will allow update firmware from Bootloader/Application mode: [ff_flasher_2021_09_24.zip](https://github.com/moonglow/FlashForge_Marlin/files/7227700/ff_flasher_2021_09_24.zip)
> 
> So, now you can choose firmware from Release page and install Marlin or FF Inventor firmware. I will update new Marlin version soon with Z=>150 for Inventor Machine, but you can try to load and check Marlin now ( do not forget what Z-MAX sets to 155 not to 150 for now )
> 
> > how do we print to it?
> 
> I use Cura for that, but you can use _ANY_ slicer you want , because Marlin is most popular firmware for 3D printers...

I was able to get FF Flasher working this time around, so thank you for that. Unfortunately, however, none of the firmwares flashed correctly and ended up with the printer in a bootloop. It should be mentioned that I was able to flash back to original firmware via my STLink on the debug port. The Flashforge firmware did work via FF Flasher, though. The original firmware did not. I wonder why that is.

As for the firmware you wrote, what's the difference between the normal bin, the swap bin, the la bin, and the swap la bin?

moonglow, 2021-09-25:
> The Flashforge firmware did work via FF Flasher, though. The original firmware did not.

You need to write frimware via bootloader mode in this case. Start flasher and than just turn on your printer, it works for me with your bootloader and i'm able to uodate printer or revert it to stock. Did you try it from bootmode ?

> what's the difference between the normal bin, the swap bin, the la bin, and the swap la bin?

LA - LinearAdvance enabled ( Marlin flow pressure control fiature )
SWAP - left and right exteuder swapped, it useful because by default FF guys makes.left extruder as 1 and right as 0 and it ia logicaly not very comforable for me.

wastelandgunner, 2021-09-25:
> You need to write frimware via bootloader mode in this case. Start flasher and than just turn on your printer, it works for me with your bootloader and i'm able to uodate printer or revert it to stock. Did you try it from bootmode ?

I was only able to get FF Flasher to detect the printer from bootloader mode. Maybe Linux lied to me as to what the normal operation PID/VID values were lol. That may possibly be the case as within Windows it reports a PID of 0x0005. Could the PID have been changed when the printer is flashed with the Flashforge firmware?

moonglow, 2021-09-25:
>That may possibly be the case as within Windows it reports a PID of 0x0005 

Looks like you run FlashForge firmware, if you will flash from bootloader mode it will be ( VID/PID ): `PID: 0x00f8, VID: 0x2b71` as you tald me before, i run your dump and i see that it is correct. My FF tools works with it too ( but you make sure what you have installed drivers in your windows system before using FF Flasher GUI ).

So now i can easily run Marlin firmware on from Monoprice firmware and update it via FF Flasher too ( only from bootloader mode ) you can try to use it ( btw if you try to use FF Flasher GUI from Linux as example WINE emulation it will not works, for linux you can try to run `ff_flash_firmware.py` if it will failed just rerun it several times and all be OK ^_^

wastelandgunner, 2021-09-26:
Update:

After modifying and using ff_flash_firmware.py with the Inventor's vendor ID and running it on Linux, I was able to get a couple different firmwares to successfully flash and run. However, the MKS firmware will result in a bootloop no matter which variant you use. Maybe this is improper usage by myself, but the Color and Classic firmwares flashed successfully.

moonglow, 2021-09-26:
> However, the MKS firmware will result in a bootloop no matter which variant you use.

To use MKS firmware you need to do some prepare steps. MKS needs very cheap external SPI flash to hold all picture and fonts for GUI, read more here: [MKS GUIDE](https://github.com/moonglow/FlashForge_Marlin/wiki/MKS-UI-based-firmware-installation-guide), so it will works too, but you need to attach top printer external SPI flash

AskAlice, 2021-11-24:
> > I have heard people have uploaded the original FF Inventor firmware onto the Monoprice Inventor to be able to use the normal FF Slice instead of the MP one. I think I might be able to find the firmware file somewhere on my computer. As for the slice it seems like its a rebranded FF slicer. [reddit.com/r/FlashForge/comments/fvoj7z/monoprice_flashforge_inventor_1_and_octoprint_and/fo5sdv6?utm_source=share&utm_medium=web2x&context=3](https://www.reddit.com/r/FlashForge/comments/fvoj7z/monoprice_flashforge_inventor_1_and_octoprint_and/fo5sdv6?utm_source=share&utm_medium=web2x&context=3)
> 
> I did attempt to do this as well before coming here. I couldn't get Flashprint to even recognize the firmware unfortunately. As for the original firmware, I may have a copy of it on a flash drive at home. I'll have to check when I get home from work.

I used flashprint 3.23.3

AskAlice, 2022-01-02:
oh yeah for flashing i tried the ff flasher on windows first, then I tried the python script in linux and both seemed to have the same issue.

Is the order of operations start script, turn on printer, wait? or start script, enable firmware update mode in settings, then wait ?

It fails to write a file or something iirc

moonglow, 2022-01-05:
@AskAlice 

> tried the python script in linux and both seemed to have the same issue.

Please provide me output log for Linux script ( make sure what you have installed FlashPrint ( and it can connect to standart firmware to make sure what all is OK with connection itself ) before it will installs also udev rules to access USB device without admin rights )

>Is the order of operations start script, turn on printer, wait? or start script, enable firmware update mode in settings, then wait ?

- Start script with firmware path as argument
- Turn on printer and wait for result

>Should I be putting inventor.bin on my internal sd card?

As universal solution about filename you can try to use firmware trigger mode trick, put same firmware with names `mp.bin` and `inventor.bin` to you `sys` folder of internal SD card and start that script what will just tell to the printer bootloader what it must be updated ( same sequence as for default script ) [ff_trigger_only.zip](https://github.com/moonglow/FlashForge_Marlin/files/7817640/ff_trigger_only.zip)

Script will output something like that:

```
python3 ff_trigger_only.py 
Searching for Flashforge printers ...
Found printer:

 DEVICE ID 0315:000a on Bus 005 Address 017 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x0315
 idProduct              : 0x000a
 bcdDevice              :  0x200 Device 2.0
 iManufacturer          :    0x1 Flashforge
 iProduct               :    0x2 FlashForge Dreamer NX 3D Printer
 iSerialNumber          :    0x3 00000000050C
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 100 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x2e (46 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0xc0 Self Powered
   bMaxPower            :   0x32 (100 mA)
    INTERFACE 0: Vendor Specific ===========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x4
     bInterfaceClass    :   0xff Vendor Specific
     bInterfaceSubClass :   0xff
     bInterfaceProtocol :   0xff
     iInterface         :    0x0 
      ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
      ENDPOINT 0x83: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
      ENDPOINT 0x3: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
Initiating control loop...
Starting control loop...
CMD M601 Received.
Control Success.
ok

Triggering firmware...
CMD M600 Received.

succeed
ok

Ending control loop...
CMD M600 Received.

succeed
ok
```

Very strange thing that it works OK for some users and some users has a problem... i think it looks like we have some root case what need to be fixed