moonglow/FlashForge_Marlin #59: Not enough X-axis movement ?

exxosuk, 2021-09-13:
I seem to have run into a really odd problem. The left extruder (T0) can move to the leftmost side of the bed at X-113. But the right extruder (T1) will only move to X-79 ?

Technically the X-axis needs to go to 113 + 34 (extruder offset) . I cannot see any gcodes for this, so can only assume it is in the firmware somewhere ?

I have tried messing about with the extruder offsets, but it only seems to alter the numbers on the screen not the physical location of the extruders.

exxosuk, 2021-09-13:
Ahh at least I can turn off the software endstop  https://marlinfw.org/docs/gcode/M211.html

moonglow, 2021-09-13:
Did you try to set active extruder before move ? ( i have one head machine ) but looks like 
```
T0
T1
```
commands will switch active extruder and than you can try to move it.

exxosuk, 2021-09-13:
Yes I did that.  When I switch, the head moves to the right 34. Which is 34 in the wrong direction. 

Oddly M211 works first time. But when I home and try again, the printer tries to destroy itself. I don't get why the first time it works.

exxosuk, 2021-09-13:
Oddly my script works fine with Pronterface.. So must be something in simply3d causing it...Sorry for wasting everyone's time.

closed 2021-09-13 by exxosuk