moonglow/FlashForge_Marlin #61: Commands not working ?

exxosuk, 2021-09-17:
I honestly don't know if I am doing something wrong, or my printer is busted, or there is a problem in firmware.

maybe someone can try this out ?

`M204 P800
M218 T1 X0 Y0
M218 T0 X-34 Y0
M907 X0.5 Y0.5 Z0.5 E0.5 

G28
M90
M302 S150
M92 X88.91 Y88.91 Z400.00 E48

M109 S200 T1
M109 S200 T0

M7
G1 X-81 Y-53 Z50 F3300

T1
G1 E50 F1000
T0
G1 E50 F1000`

what it does in my printer is it homes correctly, heats up both extruders, then moves to the front left of the printer, it then seems to change tools only, and does not exclude any filament on either extruder.

But the thing is, when I actually print a gcode file, it is otherwise works fine.

But the reason I mention this as it intermittently seems to retract the extruder rather than extrude. But this seems to happen at total random and not that often.

I have noticed something odd also when I was trying to do the "end music".  Where it will not run the entire song. In fact firstly it seems to run thing like 10 lines (out of like 100+) and plays. The second time I try, seems to only do 5 lines.. Then it just stops working completely.

It gives the impression there is some form of buffer overrun or something. Where sending commands to fast results in its some commands and ending up with a printer in some bad state.

Printing objects from whatever slicer seems to work fine. Only seems be when I send commands myself that start going nuts. But I really do not know where to post this problem because I'm not really sure where the problem is :(

exxosuk, 2021-09-17:
so this is odd, E50 extruded a little bit of filament, then does not do anything, however it I then put a high number in like E60, it then extrudes like 10mm... I think I'm doing something dumb here ?!

exxosuk, 2021-09-17:
OK nevermind then, I need to set G91 first :-\

closed 2021-09-17 by exxosuk

moonglow, 2021-09-20:
>But the reason I mention this as it intermittently seems to retract the extruder rather than extrude.

If one of extruder do retract vs extrude you need also define `#define FF_EXTRUDER_SWAP` in your config to change extruder motor direction.