moonglow/FlashForge_Marlin #67: Dremel 3D20 - Bed leveling procedure is flipped

bbatchelder, 2021-09-20:
Hey folks - so far I'm loving this firmware, thanks so much for the effort to make this happen.

One thing I noticed when leveling the bed, is that it places the first point towards the front of the bed in the center, and then two points in the rear, to the left/right.  Basically a triangle with the apex at the front and the base in the rear.  This is flipped from how it should be - as the 3D20's adjustment screws are two in the front and one in the back.

It looks like this is configured in Marlin\Marlin\BedLevelingScript.h, and I'd just need to add an if block for machine type of FF_DREMEL_3D20_MACHINE.

Just wanted to get some feedback on that before I go making any changes.

Thanks!
--b

moonglow, 2021-09-21:
You are right !  They changed triangle orientation for Dremel... If you can change it and test new points i will accept it. Thnax 

p.s: to debug new script faster, better to put all GCode commands to separate *.gcode file and run it first from SDCARD

Blitzpwnage, 2022-03-01:
Hi there, I am fairly new and just saw that the orientation is flipped. Has this been fixed? If not what can I provide to help? It's driving me crazy leveling my bed!

closed 2022-04-24 by moonglow