moonglow/FlashForge_Marlin #70: Temperature probing issues

exxosuk, 2021-09-23:
I think this is actually a problem with the firmware ?

I had the right extruder motor disconnected and it prevented the temperature probe from working. When I plugged the motor in again, the problem went away. I repeated this several times just to make sure I did not have a bad connection somewhere.

There also seems to be something a little bit too sensitive for monitoring the bed temperature ? I have my bed set at 95c. and it seems to vary between 94c -97c. But intermittently the printer shuts down saying " thermal runaway" for the bed. At this point I can only assume that if the bed temperature is about 5c more than it should be , the printer shuts down ?

moonglow, 2021-09-23:
> had the right extruder motor disconnected and it prevented the temperature probe from working. 

Bed temperature sensor ?

Bed temperature is uses ON/OFF control only ( not PID ) i think it is OK situation what you have such temperature fluctuation for 90+ i think you need to adjust RUNAWAY settings for BED to something like that => `Configuration_adv.h` 

```C
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD        60 // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS     5 // Degrees Celsius

  /**
   * As described above, except for the bed (M140/M190/M303).
   */
  #define WATCH_BED_TEMP_PERIOD                180 // Seconds
  #define WATCH_BED_TEMP_INCREASE               2 // Degrees Celsius
#endif
```

exxosuk, 2021-09-23:
Thanks. So the bed would have to be consistently at a high temperature then. I will try and check probe it has. I cannot look now as I am in the middle of a 8 hour print.

exxosuk, 2021-09-27:
Its hard to see the sensor without taking the bed apart. It seems to measure about 4.7K at room temp if that helps ?

moonglow, 2021-09-29:
Nope, sorry. First of all you need to measure it without connection to motherboard it must be about 100K, not 4,7K. Another thing different models of NTC resistors can has different temp characteristics and must be compensated over it. If we do not know exactly which resistor is used we will has temperature error. In our case i select it as default NTC ( Marlin temp sensor => 1 ). Maybe FF support will tell us more ( but they never ask before ^_^ ). Maybe i will reverse engineering original firmware more to get compensation table from it... 

p.s: you can made your own compensation table top with self measuring real temp over displayed.

exxosuk, 2021-09-29:
FF support is a total waste of time. I contacted them a few years ago asking for the MOSFET part number on the board because I had one die. They had no clue what I was even talking about never mind obtaining a part number. I managed to get the number off the part eventually and find a equivalent.

My flash forge board went totally nuts :( nothing would save to EEPROM (said it didn't have one) . The bed temperature was bearing a couple of points causing it to never stabilise which was the subject of this post. The next day the extruder temperature was jumping between 200-350 at random so I could never get the printer to even start to print. I had already changed the probes and most of the head assembly because of weird issues like this.

So I spent a couple of days fitting a Smoothie board. Was doing some test runs and set the current that 0.3amps.. Or so I thought.. I opened the topic on their own forum about this bug as usual nobody has a clue. I think it was actually pushing 3amps into the stepper motors. Originally the firmware was set at 1.50! but after sat puzzling over the burning smell.. There was a loud bang and the Z-stepper motor dropped out of the bottom of the case. It had got hot enough for the screws to melt the plastic :(

So for everyone's enjoyment..

[image]

So I think I just give up with this printer :(

I do thank you for your support during this fiasco. If you have a donation page which accepts something other than PayPal, I will gladly send you a few beers.

closed 2021-09-29 by exxosuk

moonglow, 2021-10-04:
@exxosuk 

>My flash forge board went totally nuts :( nothing would save to EEPROM (said it didn't have one) 

If you use Marlin firmware, as EEPROM i use external ( BIG ) SD Card to make sure what we will do not erase original firmware settings and no lost printer usage statistics and so on

>So I spent a couple of days fitting a Smoothie board. Was doing some test runs and set the current that 0.3amps.. Or so I thought..

That sux, but it is common for FF, you can melt this holder on original firmware too ( just by set hi current via GUI, many users does not know what it will do )....  Max current for Z motor is 0.450A. FF support will send that holder for free i think, so you can bring it to live again, but in my case i just recovery melted plastic with peace of PCB to make this connection stronger and my printer is works again.

>If you have a donation page which accepts something other than PayPal

Thank you so much, but donation is not my main goal, you can just say `Thank you` :) i accept such currency too

exxosuk, 2021-10-04:
I never thought to ask flash forge about the part. Will give them a try.. 

Oddly I have to set the motor current to 0.05A with the smoothie board. If I said any high the motor starts getting burning hot again. So I think they really have messed up the firmware somehow. I assume it is actually 0.5A going to the motor, not 0.050, as I doubt the motor would function at that level anyway.

### Thank you for your help indeed :)