moonglow/FlashForge_Marlin #71: Dremel 3D20 / Difference between la and not la? / How to build own Marlin version with FF_Tool?

T011235, 2021-09-24:
Hi, I´m currently restoring a Dremel 3d20 and the closed source firmware drives me crazy. Thankfully I found your Project. But before I´m ready to flash the firmware I have some further questions.
1. What is the difference between the colorUI_binary with the addition of la in its name and without it.
2. How can I build my own Marlin version with further functionality like a heated chamber, more fans, or other features of marlin with the FlashForge_Firmware_Tool.

Many thanks in advance for your efforts and many thanks for all the great work already done.
Have a nice weekend. ~T

moonglow, 2021-09-25:
Hi !

1. Yep you need to encode firmware before flash it with regular way ( through Dremel bootloader as example ) and if you will write firmware via JTAG programmer you use write raw firmware without encoding.
2. > with the FlashForge_Firmware_Tool

`FlashForge firmware tool` is used only for encryption/decryption firmware. For build your own firmware as example i using VSCode + PlatformIO extension, more information and guide how to build your own firmware you can find on original Marlin repo. My is already preconfigured Marlin ( by default Marlin will not supports our printers not only because they does not have configuration, but also original marlin does not have some parts of drivers and so on to make it run on our printer, maybe in future i will push all changes to mainline Marlin repo ). My default build envelopment is Linux, so build scripts and so on made for Linux. Also you need to know how to build an configure marlin firmware to get what you want to get.

T011235, 2021-09-25:
Thank you for your fast response.
If I understood correctly, I take the Marlin from your repo, configure it like normal, encrypt it with the FF_Tool and do the update steps from the how to install on Dremel 3D20.

moonglow, 2021-09-25:
1. It was already configured by default: `default_envs = FF_F407ZG` because FlashForge and Dremel use same motherboard.
2. It was already configured too, just select your machine model in `Configuration.h` and build firmware, as example for Dremel 3D20 you need just remove comment from `#define FF_DREMEL_3D20_MACHINE`

If you use Linux as your build system ( i recommends it ) you can just use my firmware build script what will do all main steps for you, as example for Dremel machine:

`marlin_builder.sh -m dreamer`

closed 2021-10-04 by moonglow