moonglow/FlashForge_Marlin #83: Failed to set trigger

loibucket, 2021-11-30:
Failed to flash, error message below

```
Start Control
Control Success
ok
Control Success
ok
Write file done
Writing to file 0:/sys/dreamer.bin Length 564464
ok
Failed to set trigger
Stop
```

Flashforge Dreamer, single motherboard version

moonglow, 2021-11-30:
Hi @loibucket !

Thank you for report ! I think i need to find windows machine to reproduce that problem.

HDHristov, 2022-01-01:
Same issue here. It should be noted, that it tries to write way past the bootloader. It is actually connecting to a fully booted Dreamer.
Looks like it is a bootloader connection issue. My dreamer was on the latest 2.15 firmware, maybe they changed something in the bootloader? Just wild guessing here.
Also tried to downgrade to 2.14 - same outcome, but if it was a bootloader change it might be still there...

HDHristov, 2022-01-01:
@moonglow would it be possible to use FlashPrint to write the Marlin binary?

On a side note, is there a description of the different prebuilt binaries? I'm referring to the different dir_la_swap options.
Thanks!

moonglow, 2022-01-04:
@HDHristov bootloader is unchangeable part of printer firmware, it will written only once by manufacture. You can try to update firmware from bootloader mode via FlashPrint too, but i think we need to fix my updater

moonglow, 2022-01-05:
@loibucket and @HDHristov 

Can you tried updated FF Flasher ? It will also allows now to set trigger only ( it will provoke bootloader to update firmware from 'sys' folder of internal uSD card ) [LINK](https://github.com/moonglow/FlashForge_Marlin/wiki/Windows-instalation-guide)

*version from 06_01_2022
p.s: make sure what your windows system installs windows FF bootloader driver and it not looks like unknown device in device manager, because you can see that driver install for application firmware and no drivers installed for bootloader.

HDHristov, 2022-01-05:
Will check tomorrow (although I already flashed using Linux). Thanks!

HDHristov, 2022-01-06:
Still does not work. Now it only says "Wait for printer" until it times out.
I can shortly see "FlashForge 3D Printer" device appearing in the Device manager when I switch on the printer. Not sure this is the bootloader device driver we need.

moonglow, 2022-01-06:
If software can't catch the bootloader that mean what you do not install driver to bootloader device. If you will install it all be OK i think

HDHristov, 2022-01-06:
It looks like I'm missing something. Where do I get that driver from?

moonglow, 2022-01-06:
>It looks like I'm missing something. Where do I get that driver from?

[image] - it will appear before application was run
Yep... if my tool can't find a printer it means that not driver installed. You can find drivers here ( in my case ): `x:\Program Files\FlashForge\FlashPrint 5\driver\win7`
And you must be quick to catch device and try to install drivers. `Libusb` can't access USB device without driver was installed on Windows.

gordonmoore9, 2022-01-17:
@moonglow - I have had trouble uploading the latest Marlin firmware using your updated FF Flasher.

However success did eventually occurred doing the following steps.

First for the purpose of explaining here, I flashed back to the FlashForge Dreamer firmware for a defined starting point.

My PC runs Windows 11. I opened Device Manager which lists device 3D Printer - FlashForge Dreamer 3D Printer as per picture.

[image]

Next opened FF Firmware Writer and selected "dreamer_2.0.9.3_la_01102022.bin". Powered off the Dreamer.

Clicked START in FF Firmware Writer and **quickly** powered on the Dreamer (for other readers note **quick power-on is crucial**). Marlin firmware started to load as per this picture.

[image]

Upon completion the Marlin Color UI appears on the Dreamer LCD.

I noticed in Device Manager that the device 3D Printer - FlashForge Dreamer 3D Printer was gone but there is now Ports (COM & LPT - USB Serial Device (COM3) as per this picture.

[image]

OctoPrint happily connects the Dreamer now via COM3 and no problem controlling the Marlin firmware.

In Device Manager the device when connecting as FlashForge Dreamer 3D Printer has hardware ID USB\VID_2B71&PID_0001.

Under the Marlin firmware, device USB Serial Device (COM3) has hardware ID USB\VID_0483&PID_5740.

Is it correct that the device changes to USB Serial Device (COM3) for hardware ID USB\VID_0483&PID_5740?

moonglow, 2022-01-18:
>Is it correct that the device changes to USB Serial Device (COM3) for hardware ID USB\VID_0483&PID_5740?

Yep, because Marlin provides their own code to implements USB serial class client and that's VID/PID is using from Marlin driver by default. 

p.s: thanx for tips, i hope it will helps to other users

closed 2022-06-29 by moonglow