moonglow/FlashForge_Marlin #84: Inaccurate touchscreen

Fred583, 2021-12-01:
Just installed the firmware on my Dremel 3D20, and it works great, other than the touchscreen being inaccurate. In the settings, for example, I have to touch the button above the one I actually want to press. Sometimes it just presses randomly as well. Is there a setting I can't find to re-calibrate it or is it something else? Thanks again

moonglow, 2021-12-01:
Hi @Fred583 !

You can recalibrate touchscreen by push for a while on free space of main screen, calibration screen will appear. You can also to run calibration utility via G-Code M995

Fred583, 2021-12-01:
Ah that makes sense, thanks! Only issue I'm having now is it let's me touch bottom left and top right, but when I click bottom right it says "calibration failed" every time

moonglow, 2021-12-01:
Marlin calibration process is tricky sometimes, i do not recommends you remove your stylus ( or whatever you use ) imminently.

1. Push to +
2. Wait while + goes to next location
3. Remove your finger or stylus and goes to next location
4. Repeat

Fred583, 2021-12-01:
Whenever I press it goes to the next + instantly, even holding for 2 or 3 seconds longer still presents the error at bottom right. It's a very interesting issue

moonglow, 2021-12-01:
Hm... i do not have Dremel 3D20 on my hands, but i will try to find solution, maybe need to tweak some sensitive related settings or something like that... Dremel 3D20 good machine, so i think modern Marlin firmware will gives new life to it, but such thing will be annoying

gordonmoore9, 2022-02-01:
I have a FF Dreamer - Could not calibrate the touch screen as described in first post.

Kept looking at Configuration.h to figure out what is preventing touch screen calibration. Turns out the answer is simple, in hindsight that is!!!!

//#define TOUCH_ORIENTATION TOUCH_LANDSCAPE was commented out. I removed // and compiled a new firmware version.

Immediately after flashing, the TFT screen went straight to calibration mode. Touched each of the 4  + and the calibration was then stored. Result! :)

moonglow, 2022-02-01:
Hi @gordonmoore9  !

Can you provide me calibration constants from calibration output ( you can invoke calibration through serial port via `M995` command, than it finished it will writes out calibration result ), i will add such solution, but very strange that it works for some users before... maybe it was too inaccurate too...

gordonmoore9, 2022-02-01:
FlashForge Dreamer

Touch screen calibration completed
TOUCH_CALIBRATION_X 16388
TOUCH_CALIBRATION_Y -11898
TOUCH_OFFSET_X -15
TOUCH_OFFSET_Y 329
TOUCH_ORIENTATION TOUCH_LANDSCAPE
//action:notification Settings Stored

moonglow, 2022-02-01:
@gordonmoore9 
I just looked in to the deep :) actualy you no need to recompile firmware you can do same with original one... `TOUCH_ORIENTATION` is `TOUCH_LANDSCAPE` by default if you do not set it via config  

```C
// XPT2046_** Compatibility
#if !(defined(TOUCH_CALIBRATION_X) || defined(TOUCH_CALIBRATION_Y) || defined(TOUCH_OFFSET_X) || defined(TOUCH_OFFSET_Y) || defined(TOUCH_ORIENTATION))
  #if defined(XPT2046_X_CALIBRATION) && defined(XPT2046_Y_CALIBRATION) && defined(XPT2046_X_OFFSET) && defined(XPT2046_Y_OFFSET)
    #define TOUCH_CALIBRATION_X  XPT2046_X_CALIBRATION
    #define TOUCH_CALIBRATION_Y  XPT2046_Y_CALIBRATION
    #define TOUCH_OFFSET_X       XPT2046_X_OFFSET
    #define TOUCH_OFFSET_Y       XPT2046_Y_OFFSET
    #define TOUCH_ORIENTATION    TOUCH_LANDSCAPE
  #endif
#endif
```

So when you set it you use new style configuration ( i will rewrite config to new style ) which will sets `TOUCH_OFFSET_*` to zero because it not sets 
```C
ifndef TOUCH_CALIBRATION_X
  #define TOUCH_CALIBRATION_X   0
#endif
#ifndef TOUCH_CALIBRATION_Y
  #define TOUCH_CALIBRATION_Y   0
#endif
#ifndef TOUCH_OFFSET_X
  #define TOUCH_OFFSET_X        0
#endif
#ifndef TOUCH_OFFSET_Y
  #define TOUCH_OFFSET_Y        0
#endif
```

and this thing will invoke touch screen calibration after firmware was started... so you can use my default firmware and invoke yourself touch screen calibration wizard by pushing on free space on main screen ( you need touch and wait ) or invoke G-code command `M995`

gordonmoore9, 2022-02-01:
I must have had a typo somewhere in my Configuration.h, started with a fresh copy and worked through adding my changes and updates (using Visual Studio Code) which colour codes section in use or not in use. I kept an eye on Conditionals_LCD.h and the // XPT2046_** Compatibility section, which had shown that the second line #if defined(XPT2046_X_CALIBRATION) && was not met. Using the fresh configuration file the whole section coloured as in use. So an oops on my part!

Invoking touch screen calibration after firmware is written to printer is a good update.

Turning back to starting touch calibration by pressing on the TFT screen which is where this issue lies. Pressing on main screen and waiting does not do as expected.

I finally realised that the first + at Top Left was displayed for the briefest of time and then jumped to Bottom Left +.

Basically the current setup does not allow enough time for the user to remove their figure in time, so it interprets that Top Left + must have been selected and moves on!

My suggestion is forget about pressing the TFT screen but add a menu item for Touch Screen Calibration. I have now added a User Menu item to invoke M995 (with Show a confirmation dialog) which is working great.

MallocArray, 2023-08-08:
Having the same issue with v0.15.1 on a Dremel 3D20 where trying to touch an empty background area goes into calibration, but also tries to register the upper left corner immediately so there is no way to touch it properly. It does wait for you to lift your finger for the lower left corner.
I'll have to try the gcode to start the calibration, but I've only been using the SD card recently, so I'm not setup right now to send gcode with Octoprint or anything.