moonglow/FlashForge_Marlin #85: motherboard sd card formatting for inventor

AskAlice, 2021-12-08:
I have an inventor 1, just replaced the motherboard and the sd card slot was empty. I had a backup of the firmware but i don't have what type of partition it was. Is it fat32? ntfs? exfat with a 128gb drive just shows the drive as full

moonglow, 2021-12-08:
HI @AskAlice !

Inventor 1 ( and Dreamer, Dreamer NX and etc ) has two SDCard one of them located on motherboard ( uSD size format ) and it used for printer boot logo, bootloader fonts and as temporary storage while we do firmware upgrade another one located at printer side  ( full ( aka big ) SD size format  )... both of them **has FAT32 filesystem format**. But please make sure that original FF firmware has a bug that shows negative size of SD card what has a size > 32GB but cards works OK in this case. So in few words you need to format your card to *FAT32*

AskAlice, 2021-12-08:
hey @moonglow thanks for your help! Do you happen to have the contents of the internal SD card?

My backup is semi-corrupted, the boot image is a white screen, but it gets into the stock firmware okay after that

AskAlice, 2021-12-08:
it shows a card error on one, both are fat32, the internal is 32gb, and external is 128gb fat32 (had to use easeus to not use exfat)               

also  i think it would be good to document what the _la _swap, and Lswap_la variants are for each build
[image]

AskAlice, 2021-12-08:
[Monoprice+Flashforge Inventor 1 board sd card.zip](https://github.com/moonglow/FlashForge_Marlin/files/7678496/Monoprice%2BFlashforge.Inventor.1.board.sd.card.zip)

here's what is on my zip

AskAlice, 2021-12-08:
when i insert the sd card internally it wipes itself, regardless of contents

AskAlice, 2021-12-08:
the motherboard is brand new, my serial port broke on the old board so i replaced it

moonglow, 2021-12-08:
_About firmware naming_
Yeh you are right i need to add such information to WIKI, prefix description:

**swap** -  by default left extruder has index 1 and right one is 0 ( original FF logic ) ... but it illogical for many users and i added that kind of build, all `swap` builds has left extruder index is 0 and right one is 1
**la** - build with Marlin [LINEAR ADVANCE](https://marlinfw.org/docs/features/lin_advance.html) feature enabled 
**dir** - some old Dreamer motherboard has inversion for right extruder and if your right extruder goes to wrong direction you need such firmware version

all other firmware is combination of features described before, as example:
`dreamer`_`2.0.9.2`_`dir`_`swap`_`la`_`11252021`.bin

Wheres:
`dreamer` - machine name
`2.0.9.2` - Marlin version
`dir` - right extruder direction is inverted
`swap` - extruders indexes was swapped 
`la` - LINEAR ADVANCE was enabled
`11252021` - build date

----
_About my FF flasher tool_ 

Looks like i need to do some refactoring to it, because many users already has problem with it ( before my last updates all was OK... and thats why it looks strange for me ) as a fallback solution i recomends your my python updater `ff_flash_firmware.py` ( to update or rever back to original firmware ) under Linux OS or you can update Marlin firmware from original FlashPrint by changing original one firmware with my firmware. I will update my FF tools soon to make sure what all users will be happy ^_^ Can you provide me your PC info ? ( OS version , x86 or x64 ).

About your backup folder ```Monoprice+Flashforge Inventor 1 board sd card.zip``` file structure is OK, but please make notes that you need to update your firmware from bootloader mode only because it is most safest way to do anything with your printer ( run script or press start AND ONLY THEN you need to turn-on your printer  after bootloader was start my tool or script will try to attach to it ). 

```https://emu.sh/flashforge-alt.zip``` - Yep thats files from LINUX based FF printers series such as Inventor 2, Adventure and so on

---
_Misc_

>the motherboard is brand new, my serial port broke on the old board so i replaced it

Yeh, common thing for FF, but it can be repaired too with the motherboard is really nice designed and will lives forever if you have some soldering skills and some IC parts

>is the partition table gpt or mbr?

Only MBR partition scheme must be used

About your videos and booting. First of all if you saw **white screen** that means that bootloader can't read boot logo ( 'sys/start.bmp' from card root ) and so on ( internal SD card is also mandatory for firmware update process, i did not recommends you to use SDHC or SDXC card as internal one, can you find some old cards for that ? as example 4 or 8 or 16GB ?  Looks like thats main reason that provides all other side effects...  And yep  ST-LINK or other JTAG programmer can solve any problem with firmware update/restore/backup :) but i prefer software way first 

### _Let's summarize_
I think you need to fix white screen issue while booting and it will helps to solve other problems or give your next step, next you will try to update firmware from bootloader mode and all be OK you will be able to changes all firmwares you wants ( stock to Marlin, stock to FF and vice versa ) just in few clicks as it was designed for.

---
p.s:  my describes is not a best and looks cryptic ,but it only for first time

AskAlice, 2021-12-08:
you did a great job of explaining this.

I dual boot windows (fully) or pop! os + windows in a VM, with gpu passthrough

What is bootloader mode? How will I get there?

moonglow, 2021-12-09:
>What is bootloader mode? How will I get there?

Printer program inside the MCU looks like that:

|STM32F407VG  MEMORY|MEM ADDRESS| CONTENT |
| :---        |    :----:   |  ---:|
| | 0x08000000       |  Bootloader firmware|
| | 0x08010000        | Application firmware |

MCU will start from `0x08000000` and firstly invoke USB bootloader it will runs, init TFT display, load fonts data from internal SD card `sys` folder, will load `boot.bmp` and show it to display and it will wait a little for `G-Code` ( i called it `bootloader mode` ) commands `~M601 S1` aka `Login` if no commands was sent it will start application firmware located at `0x08010000` and it runs and you will see TFT menu screen and `beep` music :) So to update firmware via Bootloader mode you need to connect your printer via USB to your machine, start update and next just power-on your printer to allow it be catches via my tools or other.

Update procedure:
- Update firmware via FlashPrint ( only from original firmware or bootloader mode )
- Update via my tools ( i recommends to use it only via bootloader mode )

I think firstly you need to fix your internal SD card problem to allow bootloader to shows boot logo and so on.

AskAlice, 2021-12-09:
Ahh yes That off-tune music that i hate!!!  I am getting a jtag usb toy so hopefully that will help me go deeper into this but heck yeah thanks for all the info

AskAlice, 2022-05-05:
oem33.inf
[image]

moonglow, 2022-05-08:
@AskAlice thanks for information!

Did you use linux python script or windows FF writer tool ?

AskAlice, 2022-05-08:
the python script works, windows does not when that device is used.
works on windows when the flashforge firmware you posted on here at some point is flashed to the device