moonglow/FlashForge_Marlin #86: MKS UI not working on Dreamer

scotty-sw, 2021-12-21:
Hi, before my problem I would like to say your work is amazing and much appreciated.

So I am trying to install MKS UI on my Dreamer but am unable to get it to work, I have Colour UI working using flash writer with no problems if that makes a difference. After the MKS install the screen bootloops the start bmp.

I have connected W25Q128 with dupont wires to the wifi socket but am not 100% if it is done correctly as the chip overheats after the firmware install. I followed the pinouts as per the link and connected the first SPI flash module based on numbers on the back of the WIFI module I disconnected, so if looking at the coreboard (REVH) with the writing in the correct orientation pin 1 was top left with 2 bottom left, then 3 next to 1 etc. My Dreamer only has 10 pins for the wifi so I went with that configuration. When connected like this the SPI Flash light doesn't come on during the process if that makes a difference.

Also when putting the assets onto the SD card I have been doing this on the internal SD card, do I put it in the main SD card directory or in the SRC folder with the firmware and start bmp.

When adding the asset do I put the assets folder into the directory or just the files contained in the assets folder.

Thanks in advance, any help is appreciated.

moonglow, 2021-12-21:
Hi @scotty-sw !

Assets must be located at root of your main ( BIG ) SDCard.

>When connected like this the SPI Flash light doesn't come on

I think that's mean what your pin connection is not correct, can you provide me photo of module connection ? i will try to check it

scotty-sw, 2021-12-21:
[image]
[image]
[image]

Hi moonglow, thanks for the help images as requested, also do I put the assets folder on the big SD card or just the files inside the folder

moonglow, 2021-12-21:
Oh.... looks like you do mistake in socket row... if your SPI flash is not damaged all must be OK, just recheck all pins with this ( red dot is pin 1 ) i also rotate original schematics from WIKI to make it looks from that pov

[image]

scotty-sw, 2021-12-21:
Hi Moonglow thanks again, the light on the SPI flash is now working, now the screen boots up with some square boxes that say no data, looks like it isn't doing the file transfer?

moonglow, 2021-12-21:
You need to install `assets` that was comes with firmware, just put `assets` folder to root of your external FullSize SDCARD ( it will plug from printer side and must be preformatted to FAT32 ) and turn-on your printer.... firmware will founds that dir and will install needed data from it (  if your SPI flash is OK  ) you will see messages on black screen then it was finished you will see MKS UI

scotty-sw, 2021-12-21:
Thanks heaps moonglow, it's now working, does the buy you a coffee accept paypal, will definitely do it if it does, Thanks again

moonglow, 2021-12-21:
!) Nice... btw BuyMeACoffee service is not working anymore because they drops PayPal support ^_^ i will change to something else soon ... Tnax

scotty-sw, 2021-12-21:
Okay will check back to see if you find something else, Cheers

closed 2021-12-27 by moonglow