moonglow/FlashForge_Marlin #90: PowerSpec Ultra 3D - Reboot cycle after firmware update

bgreat, 2022-02-11:
I decided to update my working PowerSpec Ultra 3D from the current Dreamer v2.15.20200917 to FlashForge_Marlin dreamer_2.0.9.3_01102022.

FF Firmware Writer did not want to work for updating the original firmware. I was able to perform the firmware update via the FlashPrint 5 Tools, Update Firmware menu. The firmware send was successful and showed success in writing the image on reboot of the printer. After the update, the initial boot screen was displayed, followed by a Marlin splash screen (all excited). Then after about 10 seconds, the controller rebooted, returned to the boot splash, then to the Marlin splash screen, and the cycle repeats... Power on/off or reset has no impact.

To recover, the FlashPrint firmware update would not work, but I was able to use FF FIrmware Write to restore the prior firmware from backup.

I did find two methods that worked to restore the prior firmware:
  1. Copy the old firmware to sys/dreamer.bin on printer internal SD, then use FF Firmware Writer to trigger the firmware update.
  2. Use FF Firmware Writer to download and trigger the firmware update to the prior version over USB. Interesting that FFFW works to restore the old version when stuck with a bad load of the new firmware.
 
So, I'm back to running the Dreamer firmware on my PowerSpec Ultra 3D. 

My PowerSpec Ultra 3D configuration:
- Model No: PS3D-ULTRA
- Purchase: Mar-2016
- FlashForge CoreBoard Rev J, V 2.2
- FlashForge DriverBoard Rev E, V 2.3
- Firmware: Dreamer 2.15 20200917

I typically work with ABS, PETG, nylon, and TPU on a glass bed w/ Kapton tape (no bonding agents).

Any guidance for how I may assist in getting this to work on my is welcome.

Enjoy!
Bill

bgreat, 2022-02-12:
I tried the prior pre-built Marlin versions and experience the same issue.

Doing some more testing, I found I can download my original firmware and revert using FlashPrint if I the timing correct - bootloader must be active - after boot splash, but before Marlin splash dislays.

FF Firmware Writer also works for the initial firmware load and reverting the firmware if I can time for the bootloader window.

bgreat, 2022-02-12:
For reference
[image]
[image]
[image]
[image]

moonglow, 2022-02-15:
Hi @bgreat !

Wow ! Thanx for so detailed report.... 

> After the update, the initial boot screen was displayed, followed by a Marlin splash screen (all excited). Then after about 10 seconds, the controller rebooted, returned to the boot splash, then to the Marlin splash screen, and the cycle repeats... 

Can you attach USB to PC ( it must appear as virtual serial port ) and attach terminal like [TerraTerm](https://osdn.net/projects/ttssh2/releases/) it can reconnect device after it was disconnected and connected again. Maybe Marlin will out some useful information. Reboots appear for many cases, but usually it means what internal watchdog was triggered and reboot the board. Can you also tries another firmware as example build for one head NX ? Another thing can you remove or insert external ( BIG ) SDCard before run the firmware.

bgreat, 2022-02-15:
Thanks for the reply! My thought was the watchdog was triggering and causing the reboot. I had not had time to set up the correct dev tools and review the Marlin code base to see if I could track down the issue. 

I'll attach as USB serial and see if any info is returned. I was considering trying to monitor, but got caught up in my "real" day job work before I could attempt to connect. I will also try the NX build to see what occurs. I normally have an external SD inserted. I believe I tried both with and without, but will confirm.

I am currently out of town for work and will not return until late next week, but will definitely update you on what I am able to find as soon as I have access to the printer. I'll also look into setting up my dev environment for Marlin to see if I can assist in isolating the source of issue.

Great work on getting Marlin working with these 3d printers and I look forward to being able assist in making this functional with my hardware.

Enjoy!
Bill

moonglow, 2022-02-16:
@OrangeBannana uses firmware on his PowerSpec Ultra #75 .Maybe he can test latest build too...  

>I am currently out of town for work and will not return until late next week, 

Thanks ! will waits for your return

I hope we can locate the problem and fix it. PowerSpaces uses Dreamer motherboards and thats why it strange for me what it not loaded correctly.... BTW if you can [dump your full MCU memory ](https://github.com/moonglow/FlashForge_Marlin/wiki/Backup-printer-firmware) i can recreate such situation on one of my motherboard and fix all it locally.

moonglow, 2022-02-19:
>I did see 2 warnings output while building, but am not too concerned based on the content:

Thats OK... just means what Marlin firmware configuration headers are embeds to firmware and can be extracted via [M503](https://marlinfw.org/docs/gcode/M503.html) new feature for 2.0.9.3  . Another WARNING is ON top because thermal chamber protection it is not good idea for our printer construction, so i disable it.

bgreat, 2022-02-26:
It was the watchdog timing out.

It did not like something about my SD card. I removed it, reformatted, and now everything is working as expected.

I still need to do my first print, but I did confirm fans, heating, and extrusion are working, plus performed a bed level.

Thank you!
Bill

moonglow, 2022-02-26:
>now everything is working as expected.
Yeh! nice to read that !

closed 2022-04-24 by moonglow