moonglow/FlashForge_Marlin #102: Part cooling fan turns off with G0

pwe169, 2022-04-09:
Hi,
I have a Flashforge Dreamer that is only a couple of months old that I wanted to use with Cura hence the use of this fantastic firmware update.

All functions work apart from part cooling. I can turn the fan on and off from the lcd, I can also use M106 and M107 but as soon as I use a G0 command the heads start to move and the part cooling turns off. The lcd still says fan is on.

if I then use G28 the fan turns on again once the G28 is completed.

Any ideas?

Pete

pwe169, 2022-04-09:
I forgot to say this issue is with the precompiled latest version Color UI. I went back to 12.0 and still had the issue

pwe169, 2022-04-09:
A bit more info, the main PCB is version 0.5 2019.

I tried to go back to the stock firmware and could not get control of the fan at all then on returning to Marlin I now can't even turn them on or off so I imagine there has been something written to the firmware that is not getting changed when you flash. 
All the fans just give a little shake when activated
Is there any way to set everything back to default?

moonglow, 2022-04-09:
Hi !

First of all i want to tell you what my firmware does not change anything what can provides side effect when you goes back to stock firmware ( thats why i do not use internal eeprom to hold settings ). All must be exactly as it was before.

> All the fans just give a little shake when activated

You mean cooling fans or rear-case fans ? Can you helps them a little with finger ( be careful ) or with some item... so turn-it on and then try to kickstart it maybe they are stick a little.

>I use a G0 command the heads start to move and the part cooling turns off.

Very strange behavior, i just install  v0.14.2 on my Dreamer NX ( one head version of Dreamer machine ), my steps:

- HOME it from LCD menu
- sent `M106 S127 ` set part fan to 50%
- sent `G0 X10 Y20` to move head

and all is OK, fans works as expected. can you provide me step-by-step guide.. maybe something goes wrong with specific sequences ?

pwe169, 2022-04-11:
Hi,

Thanks for the answer. I'll give this a try when I get home. I'll also test the wiring out with a 24v PSU to make sure nothings broken - It's only 2 months old so should be OK but it is Flashforge!!

pwe169, 2022-04-16:
Hi,
Spent today looking at what's going on. Found the issue with the part fan. There is a 2 pin connector in the head loom that is hidden away under the wrap and one of the crimps had failed! Hence when it was homed it worked and as soon as a G0 or other movements occurred the fan was not receiving any power!! Why flash forge hidden a connector under there was beyond belief!!!

My only remaining problem is the rear fans. They quiver but rarely run up to speed. 
If I set the front panel enclosure to 0 and then use M106 P1 S255 I get a faint shimmer of the fans. M106 P1 S127 gets them just about moving but no power - I can easily stop them with my finger. M106 p1 S125 seems about the best but still not much power. I'll try and get a meter on and see what's going on

Pete

pwe169, 2022-04-16:
With a meter on it the voltages are not stable as if a D to A isn't clocking out properly

moonglow, 2022-04-18:
Hi Pete!

> There is a 2 pin connector in the head loom that is hidden away under the wrap and one of the crimps had failed!

So... it was a mechanical problem ? Nice that you solve that issue 

>My only remaining problem is the rear fans.

I already hear about such problem... and if you set it to S255 they in most cases starts OK ( but sometimes you need to kick-start it by finger ), maybe i do not understand something... but they use a simple MOSFET transistor to control it and all must be simple as it can be

pwe169, 2022-04-18:
hi,
Yes you can start with your finger but the power is pretty awful compared to stock. I split the end of my finger on the rear fans with the original firmware!!

I will borrow a scope from work and see what's going on. My first guess its that the PWM isn't being setup right especially as I can't get a good reading on my multimeter - it varies all the time

Pete

moonglow, 2022-04-18:
>My first guess its that the PWM isn't being setup right especially as I can't get a good reading on my multimeter - it varies all the time

Thats a good point! Btw if you set fan speed to 255 it will set PWM to mostly logical 1 state and you can check what it have a strong power on fans. I will also check that, because maybe i missed some with new Marlin updates.

p.s: in my case S255 work for rear fans OK

pwe169, 2022-04-18:
S255 is really weak and unstable on the multimeter, s127 appears stronger - hope this helps

Pete