moonglow/FlashForge_Marlin #103: Control Failed error

TiopaSnow, 2022-04-15:
[image]
i keep getting a control failed error,

moonglow, 2022-04-15:
Hi @TiopaSnow  !

Can you provide more info ( step-by-step ) ? 

p.s:  @Mrnt provides great update to firmware update script it works very stable, but only under Linux like systems

TiopaSnow, 2022-04-15:
Downloaded and unzipped ff writer, downloaded and upzipped into the same folder the color ui, started ff writer, browsed to the nx printer files, hit start, turned on printer, printer gets past the boot loader and plays loud single note. says connected, then says connection failed

TiopaSnow, 2022-04-15:
Printer is connected to computer through usb

Mrnt, 2022-04-15:
What if you try hitting the reset button on the printer and immediately hit the Start button on the FF Firmware Write?

TiopaSnow, 2022-04-15:
same problem

TiopaSnow, 2022-04-15:
printer firmware is 1.9 20191123

TiopaSnow, 2022-04-15:
I dont have any older versions to change to either

TiopaSnow, 2022-04-15:
Found the oldest firmware, 1.2 20190320, still the same problem

Mrnt, 2022-04-15:
Please provide printer model.
Do you only have a Windows computer? No RaspberryPi, etc? Do you have Python installed on the Windows computer?

TiopaSnow, 2022-04-15:
Dreamer NX, yes just a windows pc, no python

TiopaSnow, 2022-04-15:
Do I need a raspberry pi and python? They were not listed in the instructions

Mrnt, 2022-04-15:
No you should not need them. I was asking because the alternative firmware installation script (which is primarily intended for Linux, MacOS) is a Python script. Wanted to see if you had access to a means to use the alternative method.

@moonglow could this be a code signing issue?

TiopaSnow, 2022-04-15:
I don't know if it matters but I'm in windows 7

Mrnt, 2022-04-15:
Does FlashPrint allow you to update the firmware on the printer?

TiopaSnow, 2022-04-15:
yes

Mrnt, 2022-04-15:
OOPS I updated my comment above by accident:

You are not using FlashPrint when trying to use the firmware updater?

re FlashPrint: I'm asking because I think it may be possible to use FlashPrint to send the firmware to the printer. Also to verify that there is not a PC/printer communication issue.

TiopaSnow, 2022-04-15:
I am not using flashprint paper the instructions flashprint is closed

TiopaSnow, 2022-04-15:
per*

TiopaSnow, 2022-04-15:
I got flashprint to install it, i changed the 1.2 bin file name and renamed the marlin file to its name, fflashprint installed it for me

closed 2022-04-15 by TiopaSnow

Mrnt, 2022-04-15:
Great! That's what I was going to suggest, didn't know where FlashPrint looked on Windows for the bin files.
Maybe you could provide the file path FlashPrint for Windows uses?

TiopaSnow, 2022-04-15:
C:/program files/flashforge/flashprint/firmware/

TiopaSnow, 2022-04-15:
New problem when trying to print I get: m907 overcurrent please reboot

TiopaSnow, 2022-04-15:
Do I need to tell cura it's a different printer now?

Mrnt, 2022-04-15:
did you see this:
https://github.com/moonglow/FlashForge_Marlin/wiki/%5BATTENTION%5D-Avoid-using-FlashForge-original-firmware-G-Codes

TiopaSnow, 2022-04-15:
i did now lol