moonglow/FlashForge_Marlin #104: Part Cooling Fan commands do not work after Select Tool command

mik077, 2022-04-17:
I am having an issue with GCode generated in Cura where the parts cooling fan does not turn on.  Cura inserts a Select Tool command to select the second extruder (_T1_) at the start of the file.  After this command is sent to the printer the Set Fan Speed command (_M106_) does not work if the fan index is not given.  For example _M106 S255_ does not work, but _M106 S255 P0_ does work.  There is no option in Cura that I can see to make it add the fan index to the M106 commands.

I tested this by sending _T1_ command to the printer and _M106 S255_.  The fan does move momentarily sometimes but stops.  Then when _M106 S255 P0_ is sent the fan runs as normal.

mik077, 2022-04-17:
I forgot to mention this is on a FlashForge Inventor running the precompiled firmware.

moonglow, 2022-04-19:
Hi @mik077  ! I will try to check it soon, but i has only 1 head machine... anyway i will try to check firmware logic for such case. Thanx for report.

mik077, 2022-04-19:
No problem. I switched to the firmware with the hotend swap and it's resolved my Cura issue with the part cooling fan not turning on. Great job with the FF Marlin firmware by the way.

Mrnt, 2022-04-22:
@moonglow line 2977 in configuration.h needs to be enabled:
`//#define NUM_M106_FANS 1`
I just verified this.

It looks like marlin assumes there is a part cooling fan for each head unless you specify the actual number of fans using `NUM_M106_FANS`. When you do `M106 S255` it sets the speed for the fan associated with the active extruder, so selecting `T1` means `M106 S255` will try to turn on a non existent fan. You can specify the fan by using the fan index: `M106 P0 S255`

moonglow, 2022-04-24:
Thank you  @Mrnt ! I check it right now and all works as expected, cooling fan works OK for both extruders ! Thank you !

moonglow, 2022-05-04:
@Mrnt Hm...  take a look please [HERE](https://github.com/moonglow/FlashForge_Marlin/discussions/44#discussioncomment-2689558). Looks like we need to stay FAN counts as it was and maybe create fake FAN for second extruder and make CHAMBER FAN has index 3 ?... it will looks like FAN1 and FAN2 is actually same FAN... hm need to find a logic about how does Marlin control cooling fans for two extruders.

Mrnt, 2022-05-04:
Will take a look

Mrnt, 2022-05-06:
Looks like making a second dummy fan for the same fan does not work, ie FAN0 = real cooling fan, FAN1 = FAN0, FAN3 as chamber fan - I can't get the cooling fan to turn on with `M106 P0 S255` or `M106 P1 S255` with that configuration. 
What does work is defining FAN0 only and then setting CHAMBER_AUTO_FAN_PIN to be the chamber fan - this will automatically turn on the chamber fan when it gets to 45 degrees, but M106 will no longer control the chamber fan...

Mrnt, 2022-05-06:
Reading through the bug reports in Marlin related to chamber fans it appears that `M106` is meant only for part cooling - if you want to control the other fan manually they are suggesting using `M42`. The implementation of fan control seems very convoluted and not very flexible...

moonglow, 2022-08-08:
I found a nice ( actually not ^_^ ) solution to make rear case fan works as expected and part cooling fan together, i will update such patches for next 2.0.9.5 release pack, after fix some other issue due Marlin another big source code updates :)

moonglow, 2022-10-03:
@mik077 and @Mrnt you can now try latest builds ( 0.15.0 ) now all works as expected ( some fixes was applied to Marlin code base ).

commit 8af59641 (2022-04-24): fixes #104 M106 fan control issue (#107)