moonglow/FlashForge_Marlin #105: Too many small issues-personal

TiopaSnow, 2022-04-20:
How do i go back to flashforge firmware, im running into more and more problems with it, i cant get it to keep the temp, it keeps dropping to 200 instead of what ive set, cant get it to flow correctly, i would like to undo the process.

TiopaSnow, 2022-04-20:
it worked fine for a day or so, but now no print works, they dont print at all now, i cant get the filament to flow unless its through the load/unload option

moonglow, 2022-04-20:
>i cant get the filament to flow unless its through the load/unload option
> it keeps dropping to 200 instead of what ive set

I use my printer on that firmware for about a year without any problem, if you has a unstable hotend temperature PID controlling you can adjust it with PID Autotune it will calibrate PID for your current hardware. Marlin is very powerful and stable solution IMHO.

>they dont print at all now, i cant get the filament to flow unless its through the load/unload option

Maybe you start G-code is invalid ? As example it just not waits for hotend be ready ? [WIKI](https://github.com/moonglow/FlashForge_Marlin/wiki/%5BATTENTION%5D-Avoid-using-FlashForge-original-firmware-G-Codes) some FlashForge G-codes has some different meaning vs Marlin ones. As example:
`M104` - set new hotend temperature and wait ( on FlashForge firmware )  and will not waits for Marlin firmware ( Marlin will not extrude anything on cold extruder )

>How do i go back to flashforge firmware

You can use my windows FF Writer tool or more robust solution is Linux script ( 'ff flash firmware.py' ). Just write original firmware as it was for Marlin firmware.

TiopaSnow, 2022-04-20:
took apart the hotend, in my infinite wisdom i did not check that first, a tiny piece of filament had lodged itself in the feed gears, i feel dumb.

closed 2022-04-20 by TiopaSnow

moonglow, 2022-04-21:
i think so about myself 100 times per day lol :) anyway nice what you find a root problem !