moonglow/FlashForge_Marlin #108: Retracting instead of extruding?

AskAlice, 2022-05-07:
My Inventor 1 never extrudes during prints, and when I go to filament change via the TFT display, the stepper motor first moves the filament down, tap continue, then it moves it up.

moonglow, 2022-05-08:
Hm... looks like your version of printer really use inverted logic for extruders. I already handle it with current config can you try that version with `FF_DREAMER_OLD_MB` enabled ? If all be OK i will add such configuration for my regular release cycle. We already had such issue with some Dreamer printers, but never before for Inventor.

[inventor_2.0.9.3_.zip](https://github.com/moonglow/FlashForge_Marlin/files/8646752/inventor_2.0.9.3_.zip)

AskAlice, 2022-05-08:
yes this fixed my issue, thanks

AskAlice, 2022-05-10:
@moonglow can you build this for MKS?

moonglow, 2022-05-10:
Please read first: [MKS UI GUIDE](https://github.com/moonglow/FlashForge_Marlin/wiki/MKS-UI-based-firmware-installation-guide) and make sure that all is connected correctly.

Firmwares:
[inventor_2.0.9.3_dir_mks_.zip](https://github.com/moonglow/FlashForge_Marlin/files/8664234/inventor_2.0.9.3_dir_mks_.zip)

AskAlice, 2022-06-02:
can you add this variant to the release pipeline? thanks

closed 2022-06-29 by moonglow