moonglow/FlashForge_Marlin #111: Flashforge CoreBoard 12C EEPROM on same I2C bus than 9632, c'ant access...

OneMonkeyArmy, 2022-06-03:
I have tried to access the I2c EEPROM that is actually on the same bus than the 9632 (SCL and SDA) pins PB1 & PF12 and it is numbered on the PCB as u203, and the 8pin soic/sop is labeled ATMLH428... (on flashforge coreboard)

Can someone light my lanterns, because I start to waste my time...

Thanks in advance...

Daniel Massicotte

moonglow, 2022-06-17:
What you already tries to make it works ? can you provide me more information.

p.s:  it will never be supported in mainline firmware, some board has only 256 bytes of EEPROM ( new boards has 32K ) so it useless for Marlin config for old boards and it will violate main rule to make possible to revert to original firmware without ANY CHANGES... so now if user wants to back to stock firmware it will not get any kind of side effects...

moonglow, 2022-06-17:
@theweedcher please open new issue for that, do not mix with other questions please. And provide more information which printer you use, what you tried and what you get and so on... I think we will find a solution

OneMonkeyArmy, 2022-06-22:
I never thought, that after one year working like a slave on this project, I will blow a fuse, but I did. So forget about my problems, because I have solved all of them in one shot, I simply broke, to the last part, everything related to 3D printing around me... 2 huge garbage bin on wheel full of parts of parts, Page turn... There is nothing left usable, not even a simple servo motor... 3D what ???  I no longer care !!! Fuck this shit i`m out. Summer is here, I`m goin to the beach all summer! I am FREE !!

closed 2022-06-22 by OneMonkeyArmy

OneMonkeyArmy, 2022-08-06:
By the way, it was just tired and pissed off, I didn't broke nothing, or get rid on nothing, just needed along break... I am back.. for another round... !!!