moonglow/FlashForge_Marlin #115: Case Fan

TiopaSnow, 2022-07-01:
I am using Dreamer NX with the 3-1-2022 files, the case fans dont want to turn on for more than a second, and i have no control over them with the m106 command. However if nothing is printing I can use the m106 p1 command to turn them on to any value without problem, its only when something is printing that i have no control over the fans, and they dont stay on. This is causing my printer to run in almost 50C inside the case, which is causing pillowing in my prints, i have every other fan set at 100% to try to combat this, but this is a big issue for me.

TiopaSnow, 2022-07-01:
Noting thst i cant upgrade the firmware, as the FF tool no longer sees the printer, and cant use flashprint to force the software as it also does not see the printer. I could push it throguh cura if i had a .hex instead of a .bin file

TiopaSnow, 2022-07-01:
Update::I was able to install the newest version, using a linux through WSL, and that allowed the case fans to work okayish, however it did not allow the fan that cools the heatbreak to spin, so it tried to go into thermal runaway

moonglow, 2022-07-01:
@TiopaSnow Hi ! Please tell me which version you installed i will check it. I use Dreamer NX machine so it will be fully comparable.

closed 2022-10-03 by moonglow