moonglow/FlashForge_Marlin #116: Back up of flash memory 3d20

gottlebek, 2022-08-14:
Hello to all,
i played around with the dremel 3d20 and the firmware files. I use the stm32 cube programmer with a st-link from a stm32 nucleo board.
Accidental i erased the complete chip. Now i can't bring the printer back to the original or the marlin firmware.

Does someone have a complete backup of his flash memory?

moonglow, 2022-08-14:
@gottlebek Hi !

First of all you can't use unchanged backup from other printer couse bootloader has a unique signature what bind bootloader to specific motherboard... but you can try this patched bootloader without unique ID check ^_^

[dreamer_bootloader_v1.4_20161121_no_check.zip](https://github.com/moonglow/FlashForge_Marlin/files/9333688/dreamer_bootloader_v1.4_20161121_no_check.zip)

do not forget if you use **Dreamer bootloaders** you need to use **firmware for dreamer machine** or original dreamer firmware OR you need to decrypt original DREMEL firmware with dremel key and encrypt back with dreamer key :)

gottlebek, 2022-08-14:
Thank you for your fast response and the helpfuly solution.
With your bootloader and the decrypt/encrypt i can use MarlinFw on the dremel 3d20.

closed 2022-08-14 by gottlebek