moonglow/FlashForge_Marlin #117: Build Types and No Extrusion

ntlord, 2022-08-26:
Hi,

First, can you explain what the different build types mean: like dir, swap, etc.

Second, I'm having trouble where my extruder on my Dreamer is not actually extruding anything (this is after it waits for the extruder to heat up). Any ideas? I was also running into an issue at one point where the loading of filament. E.g. unload was load and load was unload.

ntlord, 2022-08-26:
Ok. I got it to work. I had to use the DIR_LA firmware. Not exactly sure what that means. I guess I have figured out from looking at the Config.h that "swap" must mean swap the extruders.

moonglow, 2022-08-27:
Hi @ntlord !

Description of firmware types is really frequent question , so i added it now to that [WIKI PAGE](https://github.com/moonglow/FlashForge_Marlin/wiki/Prebuilt-Marlin-firmware) thanx for type me in :)

p.s: do not use LA firmware if you really do not need it, because S_CURVE acceleration is disabled in such firmwares by default... S_CURVE accel gives you really smooth movement so... in your case i think DIR_SWAP or DIR be OK

closed 2022-08-28 by moonglow