moonglow/FlashForge_Marlin #118: FF Firmware Writer not working after installing Marlin

Xan125, 2022-08-28:
I was asked to install the Marlin firmware on a flashforge dreamer and went ahead with it believing we would be able to revert. Unfortunately after getting the firmware installed we realised we were in way over our heads as we have no idea how to get it to work with cura (which was our original intention) most of what we have read has gone over our heads and when trying to revert flashprint cant see the printer it shows up as an unspecified usb device and the FF writer just starts waits and stops.

Please help us revert or get it working with cura!

moonglow, 2022-08-28:
Hi @Xan125 !

- Which printer model you have ?
- Cura is working perfectly on Marlin firmware i ( and many others ) use it as our main slicer, what is your problem with Cura ?
- First and other updates you need to do via bootloader mode ( if your FF model support it ).
- You can also try our robust solution via Linux OS using [ff_flash_firmware.py](https://github.com/moonglow/FlashForge_Marlin/blob/marlin_2.0.x/ff_flash_firmware.py) script... it works in mostly all cases ( offcause if your FF printer support update via bootloader mode ).
- Hardcore mode... you can use J-link/ST-link programmer and revert back your printer easily

Xan125, 2022-08-28:
Hi thanks for the response,

The model is the Flashforge Dreamer and with cura we actually managed to figure out how to get it to heat to temperature by removing the bit from the gcode and then it starts printing but nothing comes out. We looked around for solutions but it all just went over our heads. 

We definitely jumped into this unprepared and we are just worried we won't be able to get back to having a printer we can use. 

We don't have linux either so we will take a look at J-link.