moonglow/FlashForge_Marlin #123: Any chance of this being in upstream Marlin?

qwertychouskie, 2022-10-17:
I'd be nice to see this support added to upstream Marlin, so that we can take advantage of new Marlin features as they are added, e.g. input shaping is going to be merged into the Marlin 2.1.x branch very soon.

moonglow, 2022-10-17:
Hi!

My plans is goes to port latest Marlin branch, but i have no time to port all modification to it and test for all platforms.

AmberSpartan4588, 2022-10-30:
I'm currently in the process of rewriting Moonglows original Marlin Script and configs for current Marlin version but as Moonglow has said a lot of testing for the additional features need to be tested on Dreamer as well as writing the firmware.

digital0785, 2022-10-31:
Would love to be able to get all the new stuff. As well maybe make the inventor a decent printer

qwertychouskie, 2023-07-06:
Now that this repo is tracking upstream Marlin much more directly thanks to @tracedgod, does it now maker sense to work towards upstreaming?

tracedgod, 2023-07-06:
> Now that this repo is tracking upstream Marlin much more directly thanks to @tracedgod, does it now maker sense to work towards upstreaming?

This is going to be my next goal, right now as it stands there is a lot of customization to the code that was made on our end to get the configs working right, basically I think each board will need to be separated into their own variant (at least I'm assuming this is the best route to take), and to properly port over the ADS8118 thermocouple device to sit nicely where it needs to be with the rest of Marlin's codebase. I really do think the best route would be to get our printers into upstream, Stay tuned! 😆

digital0785, 2023-07-06:
I can't wait .. i keep checking back periodically to see if theres any precompiled pop up haha. Shot in the dark but is it possible you could rewrite the steppers so we can use standard drivers? I'd love to throw some 2208 in there that and input shaping might make the inventor a good printer yet.. thank you again for all the work you guys do !

tracedgod, 2023-07-06:
> I can't wait .. i keep checking back periodically to see if theres any precompiled pop up haha. Shot in the dark but is it possible you could rewrite the steppers so we can use standard drivers? I'd [...]

I actually am running TMC2209 steppers in mine currently, all I had to do was switch out 2 of the wires on my harness going to my driver board (I wanna say it was wires 2 & 3 I swapped?) This is the [video](https://www.youtube.com/watch?v=_U-dLgB7gOs) I followed on that subject

digital0785, 2023-07-06:
> > I can't wait .. i keep checking back periodically to see if theres any precompiled pop up haha. Shot in the dark but is it possible you could rewrite the steppers so we can use standard drivers? I'd love to throw some 2208 in there that and input shaping might make the inventor a good printer yet.. thank you again for all the work you guys do !
> 
> I actually am running TMC2209 steppers in mine currently, all I had to do was switch out 2 of the wires on my harness going to my driver board for each driver (I wanna say it was wires 2 & 3 I swapped?) This is the [video](https://www.youtube.com/watch?v=_U-dLgB7gOs) I followed on that subject

Yea I was just thinking if there was a way to re code so it's software and drop in the "normal" ones that would be much easier haha. I can't solder so doing it the diy way isn't happening for me unfortunately. I thought someplace I saw that you could do it in firmware but I don't know about that ish lol