moonglow/FlashForge_Marlin #124: Firmware updater doesnt detect printer

romxuk, 2022-11-01:
I have the 3D20 on firmware 1.2, I have installed the driver on a  laptop and on a PC but the firmware updater does not detect the printer. I have tried multiple USB ports and different leads to no avail. 

the systems I tried this on are windows 10, would using an older version of windows being needed be the cause or is there some other issue

moonglow, 2022-11-02:
Hi @romxuk !

You mean what official Dremel firmware updater can't detect your printer ? I think better to ask Dremel support about it. Please check also if your OS can detect new device while you plug your printer to your laptop USB ( check also hardware device list in your OS for unknown devices )

MallocArray, 2023-07-23:
I just got a 3D20 and attempting to update it from the original firmware as well.  I downloaded some USB drivers from the Facebook Group called "Druid Firmware" but this project also has a link to the drivers.
https://github.com/moonglow/FlashForge_Marlin/files/6657518/dremel_3d20_drivers.zip

Once I installed that, the printer was detected in Windows 10 Device Manager and then when I opened the firmware updater software, the "Printer Detected" button was green instead of gray and I was able to start the stock firmware update to 2021