moonglow/FlashForge_Marlin #125: Dremel 3d20 heating issues.

JuliThing, 2022-11-10:
I upgraded my 3d20 to the v0.15.0 firmware and I'm having issues preheating the printer, it throws a heating error everytime I tried to heat it up to 200C. It would heat up to 193C and hover around that temperature before showing a heating error and intructing to reset the printer.

moonglow, 2022-11-11:
Hi @JuliThing !

Can you tries previous version `v0.14.3` ?  If you choose lower temp all is OK ?

JuliThing, 2022-11-11:
Can I downgrade the printer directly from marlin or do I need to remove the microsd card thats directly on the motherboard.

moonglow, 2022-11-11:
You need to update firmware via internal microSD card. Instruction is [HERE](https://github.com/moonglow/FlashForge_Marlin/wiki/Installing-Marlin-on-Dremel-3D20)

JuliThing, 2022-11-11:
Turns out that this might have been a calibration issue, as when I tried again to print, this time it's working perfectly fine on v0.15.0.

closed 2022-11-11 by JuliThing