moonglow/FlashForge_Marlin #126: FF Dreamer - Heating Failed: E1 (or E2) PRINTER HALTED Please Reset

gordonmoore9, 2022-11-11:
Issue with hotend heating on FF Dreamer using your pre-built v0.15.0 DREAMER_2.0.9.5_SWAP_10022022.BIN.

**First failure** - I receive "Heating Failed: E1 PRINTER HALTED Please Reset" - the same message occurs with E2.

Observation while E1 or E2 are heating which ever temp I select - say 230°C - hotend heats up to 224°C then starts to reduce temp followed by Heating Failed.

Sometimes if I set to 200°C - display reaches 194°C then reduces to 188°C and keeps cycling up and down between these two figures. Other times it stays at 200°C +/- 1°C as expected. This also occurs with E2.

On each occasion temp fails approximately 6°C below target temp.

**Second failure** - I set E2 to 190°C, temp rose to 184°C then started cycling down to 178°C and back to 184°C. I then reduced the target temp to 180°C. Actual temp reduced down 172°C, rose to 174°C, dropped to 168°C then I received a "THERMAL RUNAWAY: E2 PRINTER HALTED Please Reset"

The heated bed appears to be stable.

If I revert to v0.14.3 I can print without any issue. E1 and E2 heat to 230°C and stay stable.

[image]

moonglow, 2022-11-11:
Thanx for report ! I will think about that issue, but looks similar to #125 .

i recommend you to use `v0.14.3` prebuilt version.

gordonmoore9, 2022-11-11:
Noted and on v0.14.3 at present.

_I do not envy the task you have when updating from one version of the base Marlin to another, then adding in the extras for the FF Dreamer and other variants. Just grateful that you do undertake this task, printing via Cura and OctoPrint is a pleasure compared with the stock FF firmware and slicer. Thank you._

gordonmoore9, 2022-11-11:
@moonglow

UPDATE - I have now had a look at a few files thinking along the lines a possible different reason why v0.15.0 failed on hotend heating, looking for something less obvious. Can't explain why I chose the following file!

I compared platformio.ini from the default MarlinFirmware/Marlin with your platformio.ini moonglow/FlashForge_Marlin.

I then made some changes to your platformio.ini as follows:

extra_scripts      =
Added in
  **pre:buildroot/share/PlatformIO/scripts/configuration.py**
  pre:buildroot/share/PlatformIO/scripts/common-dependencies.py
  pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
  pre:buildroot/share/PlatformIO/scripts/preflight-checks.py
  post:buildroot/share/PlatformIO/scripts/common-dependencies-post.py

Comment out
_# monitor_flags     =_
_#  --quiet_
_#  --echo_
_#  --eol_
_#    LF_
_#  --filter_
_#    colorize_
_#  --filter_
_#    time_

Added in
**monitor_eol       = LF
monitor_echo      = yes
monitor_filters   = colorize, time, send_on_enter**

I also copied the **configuration.py** file from MarlinFirmware/Marlin to your file structure buildroot/share/PlatformIO/scripts/

Compiled the firmware, uploaded. So now on v0.15.0 with Marlin 2.0.9.5.

I have watched the printer for the last 30 minutes, E1 and E2 sitting at 230°C and the bed at 95°C and no issues.

[image]

gordonmoore9, 2022-11-12:
Further update

So I know I am not that clever - trying to print with v0.15.0 as compiled in comment just above - printing failed!

I copied the terminal readout in OctoPrint, two files are attached:

1. E2 fails to reach 230°C and shuts down.txt - failed on heating up ahead of printing a small test item.
2. E2 reaches 230°C but stutters on way up.txt - Sent M104 S230 T1 from OctoPrint terminal - not a smooth temp increase, but has stayed at 230°C for the last 10 minutes.

[E2 fails to reach 230C and shuts down.txt](https://github.com/moonglow/FlashForge_Marlin/files/9993708/E2.fails.to.reach.230C.and.shuts.down.txt)

[E2 reaches 230C but stutters on way up.txt](https://github.com/moonglow/FlashForge_Marlin/files/9993709/E2.reaches.230C.but.stutters.on.way.up.txt)

Hope these help your analysis.

tracedgod, 2022-11-19:
Same issue here. Oddly enough, it has reached higher than 230°C a few times, E2 only happens if I am trying to change/load the filament or pre-heating extruders before print.

moonglow, 2022-11-20:
Hi @tracedgod , @gordonmoore9 !

E1 extruder is OK ? ... Can you try to run PID Autotune for extruder before retest it ? maybe PID fail with default config ( my main machine is one-head Dreamer NX ) and ranout protection is triggered ?

tracedgod, 2022-11-20:
> Hi @tracedgod , @gordonmoore9 !
> 
> E1 extruder is OK ? ... Can you try to run PID Autotune for extruder before retest it ? maybe PID fail with default config ( my main machine is one-head Dreamer NX ) and ranout protection is triggered ?

Hi @moonglow, as far as I am aware it happens to both of them. I actually tried with both, Extruder 1 will usually go up to 240°C with very little problem, but Extruder 2 will always fail no matter the temperature, I tried it at 200°C and it got to 193°C then started dropping and threw the error. I'll do some PID tuning and try it again and report back.

gordonmoore9, 2022-11-21:
I ran PID Autotune on E0 and E1 - left Dreamer on and added the PID values to Configuration.h and compiled new firmware version - got interrupted so had to stop for 10 mins. Returned to printer to see "Heating failed: Enclosure, etc"

[image]

Restarted and uploaded the firmware then set E2 to 230°C - temp held for 5 minutes. Decided to PID the heated bed, updated config and loaded another firmware version.

Started a print using both nozzles for two identical parts - "Heating failed: E1" - temp reached 219°C just before failure.

tracedgod, 2022-11-21:
Just finished PID tuning. It now does not give an error when trying to heat Extruder 2, but it's still not reaching temps, seems to be floating around ``215-220°C``. When both Extruders are heating, the error will show as ``Heating Failed: E2``.

[image]

New PID values for Hotend: ``P33.99`` ``I2.15`` ``D134.59``

Please let me know if there is anything else I can provide that could help determine the issue.

gordonmoore9, 2022-11-21:
I decided to remove PID control from hotend and bed thus using bang-bang heating.

Printed very small part with E1 to 230°C - worked ok. Then for E2, again ok.

Dreamer left on to cool down, suddenly "Heating failed: Enclosure, etc" same as my previous message.

The Dreamer enclosure does not have heating, but just realised that at start-up the enclosure temp is set to 45°C - previous v0.14.3 and earlier this was always 0°C above the enclosure symbol. The lower figure of 33°C is the enclosure temp that it just happens to have reached due to printing. Clearly 33°C is well below the (wrongly set) 45°C so printer should not have an issue.

[image]

Thatindianbruh, 2022-11-25:
Reading up on this. I feel like there was a fix provided for this issue by Marlin in their`[ 2.1.1 example configuration. ](https://github.com/MarlinFirmware/Configurations/releases/tag/2.1.1)` 

It seems they've upped the time its allowed to watch for heating to happen from 20 to 40 seconds. It might be the solution we need to get past this E1 and E2 issue. 

I don't know how to compile a new firmware nor do I know much about decompiling this image to add those changes. 

@moonglow if there's any documentation you can provide on how to compile this firmware. I'd love to learn.

gordonmoore9, 2022-11-25:
I have adjusted #define WATCH_TEMP_PERIOD from 20 to 40 in Configuration_adv.h and compiled v0.15.0.

Set Hotend E1 to 230°C - temperature held.
Set Hotend E2 to 230°C whilst maintaining E1 at 230°C- temperature held.
Set Bed to 95°C whilst maintaining E1 and E2 at 230°C- temperature held.

All three have held temp for 60 minutes now and showing stable values.

**@moonglow**
You are correct that PID Tuning is the issue. I have found various reference on the MarlinFirmware/Marlin GitHub eg Issue: [BUG] "Heating Failed" after PID takes over #21661 Closed

You will be aware that in Configuration.h at the end of the PID section the entry "#define PID_FUNCTIONAL_RANGE 10" switches on PID at 10°C below the target temperature, which correlates to the issue first reported by me.

Reading the Marlin GitHub issues, I suspect the developers have made a change to how PID calculates resulting in probably insufficient heating occurring from 10°C below target resulting in a firmware panic.

At present I would take the WATCH_TEMP_PERIOD "fix" as usable (and valid option) but hope that PID calculations are refined by Marlin. There is still a clear temp dip as PID takes over but then rises and stays at target.

**@Thatindianbruh** 
To compile this firmware - go to https://marlinfw.org/ - top menu select Install - now read section Build and Install and go from there.

Once you are setup (I would recommend the PlatformIO with VSCode option) then download MoonGlow's code and go from there.

Thatindianbruh, 2022-11-26:
I ended up bricking my 3d20. Any chance I can have you upload your boot loader?

gordonmoore9, 2022-11-26:
@Thatindianbruh

I will suggest that discussing recovery from bricking your 3D20 should not appear in this issue #126. However, at the bottom of Moonglow's main page there is a link to "Installing Marlin on Dremel 3D20 ( Windows )".

tracedgod, 2022-11-27:
All,

I decided to take a shot at updating to the bugfix-2.1.x branch from Marlin in an attempt to fix this problem and was able to successfully build the firmware on said branch. It does look like this has resolved the E2 heating issue for me! If you want to build the 2.1.x firmware, clone my repo with ``git clone --recursive https://github.com/tracedgod/FlashForge_Marlin.git`` and build the same way as moonglow's repo with the ``./marlin_builder.sh`` script. Everything on my machine seems to be working as it did on 2.0.9.5, but **__YOUR MILEAGE MAY VARY__**! 

@moonglow 
If you would like, I can send a pull request from my repo to a new branch in this repo for the 2.1.x update once it's verified by a few people that it's working fine for them. I don't want to send one to the 2.0.x branch just in case something isn't right.

moonglow, 2022-11-27:
Oh guys ! you did some R&D already !) Thanx !

So.... we can adjust some setting for current version and it will solve problem for current branch and we can goes to new 2.1.x firmware where is all solved already ? I think we need combine that two actions... first of all fix 2.9.5 version for all who use it as they main firmware and next goes to fresh branch of Marlin

@tracedgod 
yeap thanx... we need to go to the new branch and i think after some other testing we can switch to it.

tracedgod, 2022-11-28:
I tried changing the same options that @gordonmoore9 changed that resolved the problem for them on the ``2.0.9.5`` branch, but still the same problem for me. Strangest part is how I can run a PID tune cycle at ``230°C`` and it will go through and hit the temps fine, but when it comes to setting the temperature itself, no matter if the temp target is set to, it will always stop ``5°C`` below the target temp and start dropping, only difference is now it takes the 40 seconds as set in the ``Configuration.h`` file instead of the 20 seconds it was previously set to. Side note: this only seems to be a problem with the ``E2`` extruder, and for consistency-sake, I am use the ``dir_swap`` postfix build.

The only change I have not tried so far is copying the ``configuration.py`` script into ``buildroot/share/PlatformIO/scripts``. I did notice that neither the Marlin ``2.0.9.5-release`` or Marlin ``2.0.x-bugfix`` source zips do __not__ have this script located inside, but newer ``2.1.x+`` versions seem to have this script included.

There is also the fact that the issue is not present on the ``bugfix-2.1.x`` branch that I was able to get ported. Perhaps something has changed deeper in the code in regards to temperatures? I am currently looking at the commits on the github page for Marlin to see if that is the case so we may patch up our ``2.0.x`` branch to resolve this issue.

> @tracedgod yeap thanx... we need to go to the new branch and i think after some other testing we can switch to it.
> 
Sounds good. Just let me know when you have a branch ready for me to send over a pull!

moonglow, 2022-12-02:
Nice backport ! I will test it and merge to our repo ! Thanx

gordonmoore9, 2022-12-09:
I have compiled **@tracedgod** current 2.0.x branch that fixes this issue and addresses issue #128.

All good on my Dreamer, using FF_EXTRUDER_SWAP, FF_FIX_EXTRUDER_FAN, PID_PARAMS_PER_HOTEND AND PIDTEMPBED.

Hotends set to 230°C and stayed stable. Only observation is temp reaches 223°C, dips to 218°C but then rises up to 230°C and stays at temp. Think the Marlin git repo in the 2.1.x branch still has something that does not smoothly hand over from Bang-Bang to PID heating, but at least it is now very usable.

I have then run PID tuning on hotends and bed. Again all good.

I will compile the **@tracedgod** current 2.1.x branch in the coming days and test that, but expect no issues.

tracedgod, 2022-12-10:
> I have compiled **@tracedgod** current 2.0.x branch that fixes this issue and addresses issue #128.
>
>
>
> All good on my Dreamer, using FF_EXTRUDER_SWAP, FF_FIX_EXTRUDER_FAN, PID_PARAMS_PER_HOTEND AND [...]

I noticed the same thing, I believe it has something to do with how the PID calculations are written in Marlin itself, as well as the handover from Bang-Bang heating like you mentioned.

> 
> I have then run PID tuning on hotends and bed. Again all good.
> 
> 
> 
> I will compile the **@tracedgod** current 2.1.x branch in the coming days and test that, but expect no issues.

Just a note on the 2.1.x branch, at the moment it has an issue with the ''PIDTEMPBED'' and ''PID_PARAMS_PER_HOTEND'' being enabled causing the same E2 error on my machine, haven't got a chance to look at it yet. Works fine when using the default shared PID and bang-bang bed though.

moonglow, 2022-12-13:
@tracedgod FlashForge hotends uses K-type thermocouple you can change it to any others K-type sensor as well 

@gordonmoore9 looks like latest changes from @tracedgod fix that issue ?

gordonmoore9, 2022-12-13:
@moonglow Agree, now fixed.

Thanks to @tracedgod for trouble shooting the issue.

digital0785, 2023-01-13:
has this been rolled in anywhere? I'm having trouble with my inventor this precompiled firmware has this issue however 2.0.9.3 doesn't have a reverse direction for the inventor that I'm seeing. I'd love one with LA enabled as well it was great for my ender :)I'm TERRIBLE with compiling things myself so sorry for being a pain.

tracedgod, 2023-01-14:
> has this been rolled in anywhere? I'm having trouble with my inventor this precompiled firmware has this issue however 2.0.9.3 doesn't have a reverse direction for the inventor that I'm seeing. I'd [...]

Hey there @digital0785, 

I don't think the builds on this repo have been updated with these patches. I went ahead and built you the firmware for the inventor with swapped directions, as well as a linear advance build. Please let me know if you have any issues with the firmware.

[inventor_2.0.9.5_01132023.zip](https://github.com/moonglow/FlashForge_Marlin/files/10416576/inventor_2.0.9.5_01132023.zip)

digital0785, 2023-01-14:
> Hey there @digital0785,
>
> I don't think the builds on this repo have been updated with these patches. I went ahead and built you the firmware for the inventor with swapped directions, as well as a [...]

THANK YOU SO MUCH! i'll give it a shot :) I'm going to be working on calibration of the printer today gotta say aside from corner buldging due to not being dialed in/ no LA on the 2.0.9.3 got the cleanest walls i've ever gotten off this machine yesterday. Love all the extra features and having a touchscreen thats actually useful during prints keep up the grate work guys!

moonglow, 2023-01-15:
Binaries was updated !
Thanx @tracedgod