moonglow/FlashForge_Marlin #128: FF Dreamer - Chamber Temp Set at 45°C

gordonmoore9, 2022-11-24:
I believe I have discovered where the chamber temp is being incorrectly set for the FF Dreamer to 45°C (I made reference to this in issue #126.)

In Configuration_adv.h there is the entry :-

 * Startup commands
 *
 * Execute certain G-code commands immediately after power-on.
 */
#define STARTUP_COMMANDS "M141 S45"

Marlin G-code is defined as "M141 - Set Chamber Temperature" hence this causes an issue.

tracedgod, 2022-11-27:
I'm pretty sure this is set this way to have the case fans kick on automatically at ``45°C``. I have played around with setting the chamber temps and when it is set lower than the current chamber temp reading (but __not__ set to ``0°C``), the fans will turn on. When the chamber temp is set, you cannot manually turn on the case fans, but when chamber is set to ``0°C``, the fans are free to be set manually either through the GUI or with G-code command ``M106 P2 S255`` (``P`` value is the fan index, in this case ``P2`` is our case fans, P0 is the hotend cooling fan, and the ``S`` value can be anywhere between ``0-255`` depending on your desired speeds). 

__**In my opinion**__, it would probably be better to move this G-code command into the Start G-code section of a slicer like Cura and remove it from the ``STARTUP_COMMANDS`` definition in ``Configuration_adv.h``. 

For example, the start G-code would probably look like this:

```
;Start Gcode
G90
M141 S45     ;Set chamber temperature for cooling fans
M140 S{material_bed_temperature_layer_0}     ;Heat bed up to first layer temperature
M190 S{material_bed_temperature_layer_0}     ;Wait for Heat Bed temperature
M104 S{material_print_temperature_layer_0}    ;Set nozzle temperature to first layer temperature
M109 S{material_print_temperature_layer_0}    ;Wait for Extruder temperature
M107 ;start with the fan off
G28
M117 Purge extruder
G1 Z50.000 F420
;Purge line
G1 X-110.00 Y-60.00 F4800
G1 Z{layer_height_0} F420
G92 E0
G1 X-110.00 Y60.00 E17,4 F1200
;Purge line end
G92 E0
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
```

gordonmoore9, 2022-11-28:
Agree with adding M141 S? to Start G-code as appropriate, depending on material requirements.

Or in Cura, use different print profiles for ABS (with no fans), PLA (with fans), etc as appropriate.

moonglow, 2022-11-28:
Chamber temp a hack to control chamber fans. We do not have heated chamber, but we need a way to control fans easily. @tracedgod is absolutely right... it is just on/off trigger for chamber fan ( and 0 temp as shutdown mode )....  Historically i sets it to 45C... but better to set it to 0 by default and control it via your favorite slicer.

tracedgod, 2022-11-28:
> Agree with adding M141 S? to Start G-code as appropriate, depending on material requirements.
> 
> Or in Cura, use different print profiles for ABS (with no fans), PLA (with fans), etc as appropriate.

I was thinking the same thing, I just need to look in Cura a bit more and see if there's an option to set chamber temps by default or there is something I need to add to my printer profile for that to appear.

closed 2022-12-13 by moonglow