moonglow/FlashForge_Marlin #131: Marlin being sent but not unpacking and applying.

pooberry, 2022-12-26:
Hi I have a Flashforge Dreamer that I purchased this year and upgraded to having a full metal hotend but can now not print at temperatures above 248 and this firmware seemed to be the best solution since I am not able to revert it back to previous firmware version due to what seems to be by design from Flashforge and the motherboard that is in the printer. 

What happens is that I run ff_flash_firmware.py while connected to the printer and reboot and nothing happens. If I send the files during boot it will says "File Write Success" and then do nothing. I've tried various boot cycling and put it back to the stock firmware with Flashprint multiple times with no success. 

I then found the internal micro SD card on the board could be removed and that I could put the firmware directly on what I thought was the systems internal storage. However it still boots back to stock rather than the previous firmware version I found that allows for printing above 248 C. And from what I found, this is due to an update process needing to be triggered to force the printer to look at its internal storage firmware.bin file and apply the file that I put in the /sys folder there. 

I'm kinda in the weeds here and I'd rather run something like Marlin that the stock firmware in a previous version but I am not sure how to get the changes to apply let me know if you are able to assist and if there is anything specific about my device that is needed.

moonglow, 2022-12-27:
Hi @pooberry :)

>full metal hotend but can now not print at temperatures above 248

You can try also my another project if your prefer original FlashForge firmware ( this is original firmware patched to allow user set higher temperature ): [THIS](https://github.com/moonglow/flashforge_fan_fix) and [RELEASE PAGE](https://github.com/moonglow/flashforge_fan_fix/releases)

About  Marlin:

>while connected to the printer and reboot and nothing happens.

You did tried prebuilt firmwares and printer just not boot at all or boot original FF firmware anyway ?

>I then found the internal micro SD card on the board could be removed and that I could put the firmware directly

Yep but printer will not use firmware without UPDATE TRIGGER command was received by bootloader.

pooberry, 2022-12-28:
> while connected to the printer and reboot and nothing happens.
It boots the original firmware. 

Thank you for the information on the original release I'll let you know how it goes.

moonglow, 2022-12-28:
Update trigger is sent by our update script. Can you provide answers for other questions? Thanks

pooberry, 2022-12-28:
About Marlin:

- I use `python3 ff_flash_firmware.py marlin_builder.sh` with success. Printer boots back to original firmware as if nothing was applied.

You did tried prebuilt firmwares and printer just not boot at all or boot original FF firmware anyway ?
- Yes. I asked Flashforge support for firmware version 
dreamer_2.4.20160603.bin and I used the path to this file as an argument with ff_flash_firmware.py but the update is not applied after boot.

I did use ff_flash_firmware.py to successfully with the current firmware file when the printer was stuck on boot which was the result when sending the firmware update while the printer was displaying the startup image.

zerog2k, 2023-03-26:
Unsure if it's related, but I have a PowerSpec Ultra3D (basically dreamer board inside). The internal sd card on this one uses a file called `power.bin` instead of Dreamer's `dreamer.bin` as the firmware it flashes on boot.
I found that the current `ff_flash_firmware.py` writes to internal sd card (M28 command) as `firmware.bin`, but the board seems to ignore this (and continue to try to reflash the existing power.bin on internal sd card). 
I was able to change the script (in my case to `TARGET_FIRMWARE_NAME = 'power.bin'`), and after machine reboot & flash, it came up with Marlin fw.
So maybe try `TARGET_FIRMWARE_NAME = 'dreamer.bin'` for real Dreamer?