moonglow/FlashForge_Marlin #135: print names cut off on 2.0.9.5

digital0785, 2023-01-14:
I updated to 2.0.9.5 ( thank you so much tracegod for the updated firmware compile) and noticed that when looking at the files on the sd it cuts it off like half way through the screen as well as when you select it to confirm. 2.0.9.3 it would show the full width on the sd contents screen and fully show the file name when selected, this helps greatly when you're doing prints at multiple different settings for things  ( when i'm testing things I generally put the settings i'm testing in the file name to keep track)

thank you again for everything this has actually made me like my inventor finally~!

tracedgod, 2023-01-17:
Strange, I actually have never run the 2.0.9.3 builds on my printer, only 2.0.9.5, so I assumed that was how marlin showed filenames.. I'll flash 2.0.9.3 to my printer to see if the full filenames show on my Dreamer and go from there.

digital0785, 2023-01-17:
I truly appreciate all your hard work and how quick you guys are to jump on things for such a niche market!.. it's really a minor inconvenience but just odd it's not present on the old ones

tracedgod, 2023-01-17:
I tried out 2.0.9.3 and sure enough the full file names show. I was able to narrow it down to a configuration flag in ``config_adv.h`` to allow long filenames and have it scroll the name too (a new feature in 2.0.9.5+ it seems). Since we are also in the process of upgrading to Marlin 2.1.x from upstream, I'm thinking it to be best to update the builds for the new Marlin release version instead of another small patch to the 2.0.9.x branch. 

@moonglow, should we wait to fix this in 2.1.x or should I setup a pull request to fix the current branch? I can always send over a pull request for the 2.1.x branch if you would rather move forward with the next release. I just built a 2.1.2 release a few minutes ago and all seems to be functioning as expected! 🤔

digital0785, 2023-01-17:
> I tried out 2.0.9.3 and sure enough the full file names show. I was able to narrow it down to a configuration flag in `config_adv.h` to allow long filenames and have it scroll the name too (a new [...]

AWESOME! I'm not worried about it for not just figured a nice QOL change. Looking forward to all the new things with upstream as well this firmware swap actually has me enjoying using the printer now.. quality of prints and speed is up HUGE

digital0785, 2023-01-18:
I know unrelated but is it possible to turn on bed PID tuning for the upstream updates as well :) I'm gonna see if i can figure it out to compile it myself. But figured didn't hurt to ask haha.

tracedgod, 2023-01-18:
> AWESOME! I'm not worried about it for not just figured a nice QOL change. Looking forward to all the new things with upstream as well this firmware swap actually has me enjoying using the printer now.. quality of prints and speed is up HUGE

Absolutely! Best upgrade you can do to these older printers to really bring them back to life!

> I know unrelated but is it possible to turn on bed PID tuning for the upstream updates as well :) I'm gonna see if i can figure it out to compile it myself. But figured didn't hurt to ask haha.

I was considering doing that along several other changes I made in a testing branch. I was able to get the internal SD card and external SD card both detected in Marlin, super experimental though, I had to hack up quite a bit of the code to get that to work! I might try to see if I can get MKS_UI to run using the internal SD for storing the files needed for that UI build instead of needing the SPI flash, not sure if would work though. Also was trying to get the WiFi working, but I accidentally killed my WiFi chip when dumping firmware... so now I'm writing a firmware for a Raspberry Pi Pico W to act in it's place, no ETA on that yet though!

digital0785, 2023-01-18:
that would all be amazing imagine running a makerbot style printer on octoprint with ease next you're going to say you got the camera working LOL keep up the great work man!